ros-perception/image_pipeline #527: Noetic Release

SteveMacenski, 2020-05-13:
Currently blocked by https://github.com/ros-perception/image_pipeline/pull/507

sloretz, 2020-05-15:
:tada: Unblocked by #530 :rocket:

SteveMacenski, 2020-05-15:
You and Jacob are keeping me busy https://github.com/ros-perception/perception_pcl/issues/278

sloretz, 2020-05-15:
> You and Jacob are keeping me busy ros-perception/perception_pcl#278

I'm happy to help if it's only the release left. If these are the right steps, with your permission I'll make it happen.

1. Update changelogs on branch `noetic` and bump `package.xml`s from `1.14.0` to `1.15.0`
1. git tag `1.15.0`
1. Create file `noetic.ignored` with content `wiki_files` in https://github.com/ros-gbp/image_pipeline-release
1. bloom release creating a track called `noetic` to rosdistro `noetic`

SteveMacenski, 2020-05-15:
Sorry, I meant that as a joke, that was probably lost in text. I'll get to it tomorrow if you don't. It was just a little late today for me to start that process by the time that was ready to go.

sloretz, 2020-05-15:
> Sorry, I meant that as a joke, that was probably lost in text

I figured, but thought I'd offer in case there was a lot on your plate at the moment.

> I'll get to it tomorrow if you don't.

I interpreted that as permission :wink: https://github.com/ros/rosdistro/pull/24924

closed 2020-05-15 by sloretz