ros-perception/image_pipeline #533: Camera Calibration package outputting incorrect YAML?

LukeAI, 2020-05-17:
I've been trying to parse the camera calibration .yaml file via the ROS parameter server ([my ROS answers question here](
https://answers.ros.org/question/352519/how-to-retrieve-list-from-ros-parameter-server-that-is-not-comma-separated/)

and I found that the output of camera_calibrator in Kinetic is like the below.

distortion_coefficients/data has no comma separators. Is this intentional or a bug?

```
image_width: 1920
image_height: 1200
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1043.889715, 0.000000, 946.091722, 0.000000, 1045.536555, 601.201155, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.034062 0.026819 0.000826 -0.002720 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1041.022827, 0.000000, 938.549637, 0.000000, 0.000000, 1045.705078, 602.181309, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]

```

SteveMacenski, 2020-05-18:
I don't think that is probably intentional, a PR to resolve this would be appreciated.

mikeferguson, 2024-01-22:
Appears fixed in https://github.com/ros-perception/image_pipeline/commit/99ebb74b4f3036ffbee0fe550220206fa5a23ba2

closed 2024-01-22 by mikeferguson