ros-perception/image_pipeline #539: Trouble running image_saver from roslaunch

mgerhardt10, 2020-05-27:
Hi!

I'm trying to run the image_saver tool from roslaunch, and I'm having some difficulty. Specifically, I'm trying to implement this [answer](https://answers.ros.org/question/9377/how-to-take-save-a-snapshot-from-my-kinect-to-disk/?answer=235294?answer=235294#post-id-235294) on the Wiki.

Here's the launch file:

```xml
<?xml version="1.0"?>

<launch>

  <!-- This is a sensor_msgs/Image -->
  <arg name="image_topic" default="/camera/rgb/image_raw" />

  <node pkg="image_view" type="image_saver" name="image_saver" >
    <remap from="image" to="$(arg image_topic)"/>
    <param name="save_all_image" value="false"/>
    <param name="filename_format" value="%04d.%s"/>
  </node>

</launch>
```

However, when I try to call the service

`rosservice call /image_saver/save`

no image is saved. Any suggestions are appreciated!

SteveMacenski, 2020-05-27:
Please ask on ROS answers and provide verification that your topic is correct and you’re setting the params correctly.

closed 2020-05-27 by SteveMacenski

mgerhardt10, 2020-05-27:
I asked on ROS Answers [first](https://answers.ros.org/question/353291/trouble-with-image_saver/) but to no avail, which is why I posted here.