ros-perception/image_pipeline #551: Saving tar fails in python 3.8

PfeifferMicha, 2020-06-24:
When pressing "save" in cameracalibrator, the tool fails to save the tarballs and prints:
```
...
/image_pipeline/camera_calibration/src/camera_calibration/calibrator.py", line 1266, in taradd
    if isinstance(buf, basestring):
NameError: name 'basestring' is not defined
```

It looks like "basestring" [is deprecated](https://docs.python.org/3.0/whatsnew/3.0.html) in python 3.8. That document recommends replacing basestring with str, however then saving still fails with

```
Traceback (most recent call last):
  File "/home/user/Projects/ROS/catkin_ws_noetic/src/image_pipeline/camera_calibration/src/camera_calibration/camera_calibrator.py", line 273, in on_mouse
    self.c.do_save()
  File "/home/user/Projects/ROS/catkin_ws_noetic/src/image_pipeline/camera_calibration/src/camera_calibration/calibrator.py", line 578, in do_save
    self.do_tarfile_save(tf) # Must be overridden in subclasses
  File "/home/user/Projects/ROS/catkin_ws_noetic/src/image_pipeline/camera_calibration/src/camera_calibration/calibrator.py", line 1279, in do_tarfile_save
    taradd('left.yaml', self.yaml("/left", self.l))
  File "/home/user/Projects/ROS/catkin_ws_noetic/src/image_pipeline/camera_calibration/src/camera_calibration/calibrator.py", line 1275, in taradd
    tf.addfile(tarinfo=ti, fileobj=s)
  File "/usr/lib/python3.8/tarfile.py", line 1995, in addfile
    copyfileobj(fileobj, self.fileobj, tarinfo.size, bufsize=bufsize)
  File "/usr/lib/python3.8/tarfile.py", line 256, in copyfileobj
    dst.write(buf)
  File "/usr/lib/python3.8/gzip.py", line 276, in write
    data = memoryview(data)
TypeError: memoryview: a bytes-like object is required, not 'str'
```

I currently do not know enough about tarballs to fix this myself, maybe someone can have a look?

Working on noetic branch, commit 98364014ae3d4aa6ca92a0c7f6e8541ad0354675

SteveMacenski, 2020-06-24:
@PfeifferMicha thanks for the note. Can you try `memoryview(bytes("This is a string", encoding='utf-8'))`? That should get you the right types. If that all works out, please submit a PR and we'd be happy to merge it in

PfeifferMicha, 2020-06-26:
Done, pull request is open. I ended up removing the dependency on StringIO entirely, as using it would give me a lot of troubles downstream. Saving now works on python3.8, Ubuntu 20.04.

Side Note: Since the current head of branch 'noetic' is broken (Charuco integration is halfway done), I made this pull request on top of the previous commit, 98364014ae3d4aa6ca92a0c7f6e8541ad0354675. See also #550.
Maybe the Charuco implementation could be moved to another (temporary) branch until it is fixed? Or is 'noetic' considered a development branch? In this case, is there a "stable" noetic branch?

SteveMacenski, 2020-06-26:
What do you mean its half implemented? @JStech

JStech, 2020-06-26:
I don't currently have a Noetic install, so I just cherry-picked the commits and trusted CI. Everything is there. I can probably look into it this weekend.

SteveMacenski, 2020-06-26:
I think we need more info from @PfeifferMicha. Functionally noetic and melodic shouldn't really vary much so I want to know what they believe is "halfway done" to know what needs updates.

PfeifferMicha, 2020-06-26:
By "halfway done" I mean I needed to change quite a few things to get it to not crash. Then once that was done I noticed Stereo Calibration (my actual goal) was not implemented yet.
I had closed #550 since I thought this was sort of intentional (with "WIP" in the commit message), I'll reopen it and give more details there.

SteveMacenski, 2020-06-26:
@JStech can you follow up on #550? Looks like your commit may have introduced some issues.

Stereo for those tags haven't been implemented yet, you're welcome to do so. Its often a pain to implement a new tag for mono/stereo/fisheye so its not expected that someone does all of them (and in some cases the library may not support them)

JStech, 2020-06-26:
Yeah, I'll check it out. "Too many values to unpack" is an error I became very familiar with in implementing this feature, so it's probably my fault. Sorry about that.

And yes, stereo and fisheye are both unimplemented with the ChArUco board at this point because OpenCV doesn't have those functions implemented for ChArUco boards, and, frankly, it's not something I need right now.

SteveMacenski, 2020-06-26:
if its not in openCV, that's enough of a reason for me

JWhitleyWork, 2020-06-29:
@JStech / @PfeifferMicha Can you please verify that #554 fixed this issue?

PfeifferMicha, 2020-07-08:
With #550 still open, I cannot verify, as (stereo) calibration fails. As soon as #561 is working and merged, I can try to verify this (for noetic, I still don't have a ros2 setup).

JWhitleyWork, 2020-07-29:
@PfeifferMicha Can you please re-test this now that #561 is merged?

mikeferguson, 2024-01-17:
Presumed to be fixed by last noted commit, since no further feedback

PR #554 (2020-06-29): Removed basestring (no longer exists in new python 3 version).
Fixes #551
commit 8af665af (2020-06-29): Removed basestring (no longer exists in new python 3 version). (#554)

Fixes #551
PR #552 (2020-06-29): Removed basestring (no longer exists in new python 3 version).
Fixes #551

Note: Only tested on Ubuntu 20.04, python 3.8.2.
commit 62e9053e (2020-06-29): Removed basestring (no longer exists in new python 3 version). (#552)

Fixes #551
closed 2024-01-17 by mikeferguson