ros-perception/image_pipeline #557: Avoid Error and return on different frame_ids

ToniRV, 2020-07-01:
Hi,

In most of the nodeletes, whenever their subscribe to two images, say for example depth and rgb image in point_cloud_xyzrgb, the `frame_id` of both images are compared and if not the same, then the node raises an error and stops.

It would be nicer if we could just raise a Warning and continue.

The case that I try to solve is when there is an RGB camera with frame id "rgb_cam" and a depth camera with frame id "depth_cam", which are already registered but have different frame_id (which I can't change for various reasons) in their header.
The node point_cloud_xyzrgb and point_cloud_xyzi will just return because of mismatching frame_ids ...

https://github.com/ros-perception/image_pipeline/blob/56c3874e55de78cc973aa693da95a605d5a6a9f4/depth_image_proc/src/nodelets/point_cloud_xyzi.cpp#L147-L153

Maybe something like this would be more user friendly:
```cpp
 if (depth_msg->header.frame_id != intensity_msg_in->header.frame_id) 
 { 
   NODELET_WARN_THROTTLE(5, "Depth image frame id [%s] doesn't match image frame id [%s]", 
                          depth_msg->header.frame_id.c_str(), intensity_msg_in->header.frame_id.c_str());
 }
```

ToniRV, 2020-07-01:
Same for point_cloud_xyzrgb:
https://github.com/ros-perception/image_pipeline/blob/56c3874e55de78cc973aa693da95a605d5a6a9f4/depth_image_proc/src/nodelets/point_cloud_xyzrgb.cpp#L162-L168

I'll be happy to open a PR if you agree on changing this! Thanks!

SteveMacenski, 2020-07-01:
Would it be better to convert into the new frame when they vary?

ToniRV, 2020-07-02:
Sorry @SteveMacenski, not sure what you meant by that.
What is the "new frame"? 

It seems like the header for the output pointcloud is just the same as the depth msg:
https://github.com/ros-perception/image_pipeline/blob/56c3874e55de78cc973aa693da95a605d5a6a9f4/depth_image_proc/src/nodelets/point_cloud_xyzrgb.cpp#L258

I would keep that, and just warn the user that we are using the depth msg as a frame_id.
We can't just modify either of the depth or rgb msg frame_id in any case as these are the const inputs to the callback.

SteveMacenski, 2020-07-02:
If you have some data in rgb frame and some in depth frame, and those frames are not coaligned, you would need to transform one into the other frame so that they can be accurately registered. The error of 1cm on the accuracy of the fusion would be huge.

ToniRV, 2020-07-02:
Yes totally agree, but I would argue that that should be done by using the RegisterNodelet? https://github.com/ros-perception/image_pipeline/blob/1138200c7d955400488559b2b33b0734f8a976fd/depth_image_proc/src/nodelets/register.cpp

My point in this issue is that: while we keep the assumption that the frames are _already_ registered, we do not oblige the users to set the **name** of the frame_id to be the same. 

And this can happen if you have two tfs: one for the depth and one for the rgb, that happen to be aligned already (aka the same tf) but you want to keep them separate for various reasons (e.g. if you only care about depth you may not want to deal with the rgb tf, but only with the depth tf).

So at the end of the day, the "fix" that I would do is just to warn the user instead of just deliberately returning without output.

SteveMacenski, 2020-07-02:
I disagree, if the frames are different, its because they're different. If they were the same, there'd be no need to distinguish them. In that case, you should file a ticket / PR with the sensor driver to change this. 

I believe the error case is the correct case to stop from unexpected results when swapping between many different cameras / vendors. 

@JWhitleyWork do you disagree with that assessment?

ToniRV, 2020-07-02:
Also agree with the point you raise, it just depends on the philosophy of the code: do we prefer to ask for permission or forgiveness?
IMHO: permissions here would be:
```cpp
if (ok_operation()){
return do_operation();
} else {
return false;
}

```

while the forgiveness one would be:
```cpp
return do_operation();
```
Both are fine honestly.

JWhitleyWork, 2020-07-29:
@ToniRV / @SteveMacenski I agree with @SteveMacenski here. If a camera driver is producing more than one image stream, there is no use-case that I can think of where they would/should be in different frames. If you can provide a specific use-case where this would be true, I'm happy to consider it. Otherwise, I consider this an error and the code is working-as-intended.

closed 2020-07-30 by ToniRV

shubham-shahh, 2022-01-21:
> I disagree, if the frames are different, its because they're different. If they were the same, there'd be no need to distinguish them. In that case, you should file a ticket / PR with the sensor driver to change this (or whatever nodelet may be assigning frames).
> 
> I believe the error case is the correct case to stop from unexpected results when swapping between many different cameras / vendors.
> 
> @JWhitleyWork do you disagree with that assessment?

Hi, but since most cams publish transforms for different frames, why are we not taking those transforms into account and solving this issue?