ros-perception/image_pipeline #572: Replace matrix subclass with ndarray in camera_calibration tests

JWhitleyWork, 2020-07-29:
In the file `camera_calibration/test/test_directed.py`, the following warning is generated when running tests (at least on `foxy`):

```
  Warning: the matrix subclass is not the recommended way to represent matrices or deal with linear algebra (see https://docs.scipy.org/doc/numpy/user/numpy-for-matlab-users.html). Please adjust your code to use regular ndarray.

-- Docs: https://docs.pytest.org/en/latest/warnings.html
```

samuel-ongzx, 2024-02-17:
Just took some time to look into this, the warning gets trigger from [L132](https://github.com/ros-perception/image_pipeline/blob/14f604a8c04c079339e43b2a5a1e3ec6dec94b29/camera_calibration/test/test_directed.py#L132) in `camera_calibration/test/test_directed.py`, specifically `assertAlmostEqual()` which is part of python `unittest` library. Since I can't change the `unittest` library, does anybody have any suggestions on how to resolve this warning?

The trace of pytest warning:
[image]

ScottMonaghan, 2024-03-23:
Taking a look a this the issue seems to originate withe the [ros-perception/image_geometry](https://github.com/ros-perception/vision_opencv/tree/rolling/image_geometry/image_geometry) package, specifically in [cameramodels.py](https://github.com/ros-perception/vision_opencv/blob/rolling/image_geometry/image_geometry/cameramodels.py):

`PinholeCamera`'s exposed members--`K`, `D`,`R`,`P`,`full_K`, and `full_P` are initialized as `numpy.matrix` objects. (see screenshots below).

I'll create an issue in [ros-perception/image_geometry](https://github.com/ros-perception/vision_opencv/tree/rolling/image_geometry/image_geometry), and go into more detail there.

[image]
...
[image]

I'll create a new issue in ``

ScottMonaghan, 2024-04-20:
The original issue has been fixed with https://github.com/ros-perception/vision_opencv/pull/527.

With the original request, this particular issue can be closed.

That said, in the updates to the PR above, new deprecation warnings were introduced (see below). 

I will create a new issue for those, and follow up with a PR with corrections.

New warnings:
```
============================== warnings summary ===============================
src/camera_calibration/calibrator.py:47
  Warning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives

test/test_directed.py::TestDirected::test_stereo
test/test_directed.py::TestDirected::test_stereo
test/test_directed.py::TestDirected::test_stereo
test/test_directed.py::TestDirected::test_stereo
  Warning: Call to deprecated method fromCameraInfo. (The fromCameraInfo() method is deprecated as of J-turtle, and will be removed in K-turtle. Please use the from_camera_info() method instead.) -- Deprecated since version J-turtle.

test/test_directed.py: 192 warnings
  Warning: Call to deprecated method projectPixelTo3d. (The projectPixelTo3d() method is deprecated as of J-turtle, and will be removed in K-turtle. Please use the project_pixel_to_3d() method instead.) -- Deprecated since version J-turtle.

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< camera_calibration [31.7s]

Summary: 1 package finished [32.1s]
  1 package had stderr output: camera_calibration

```

ScottMonaghan, 2024-04-20:
Update: the updated image_geometry 4.1.0  package hasn't been migrated to packages.ros.og as of writing, so unless you build from source the bug will persist. 

So you may want to keep this issue open for now.

I will keep an eye on the package, and once it's fully available for rolling on packages.ros.org we should be able to fully close this issue.

ahcorde, 2024-04-22:
if the package was release, we can close this issue for now. If the problem persist you can open a new issue or open this one again. 

Thank you for your contribution!

closed 2024-04-22 by ahcorde