ros-perception/image_pipeline #574: Repository Maintenance

JWhitleyWork, 2020-07-29:
Calling @SteveMacenski, @vrabaud, and @yechun1. If I've named you here, you are currently listed as a maintainer on at least one package of the `ros2` (currently `foxy`-targeted) or `noetic` branches of this repository. Please confirm that you are still able to/interested in maintaining both the ROS1 and ROS2 versions of your assigned packages. If I don't get a response from you within a month, I'll assume that you're no longer available for/interested in maintaining these packages.

@SteveMacenski: Currently listed for all ROS1 and ROS2 packages.
@vrabaud: Currently listed for all ROS1 and ROS2 packages.
@yechun1: Currently listed for the ROS2 `depth_image_proc` and `image_publisher` packages.

SteveMacenski, 2020-07-29:
I'm not particularly interested in maintaining things, but if I were to need to, ROS2 would be the place. I'm mostly stressed out about the rolling distro build failures. This is the only repo I have that is in the rolling distro and it makes me regret even having the 1. Its too much. If we get off rolling, I'd be more inclined to continue in a passive way (help as can, but I don't use this package so its hard for me to really do much more than just review PRs and close erroneous tickets).

JWhitleyWork, 2020-07-29:
Thanks, @SteveMacenski. I'm happy to remove your name/email from the maintainership list so you stop getting notifications. If you then just watch this repo and jump in when you can (or when I assign you something), that would work fine for me. Alternatively, I can leave you on the ROS2 packages and do my best to make sure that Rolling continues to build (I believe #573 should fix this for now). Your preference?

SteveMacenski, 2020-07-30:
I could do that, sure

JWhitleyWork, 2020-08-01:
Which one?

SteveMacenski, 2020-08-02:
Oh sorry, keep watching but remove from maintainers list.

JWhitleyWork, 2020-08-28:
Issues raised by @SteveMacenski have been handled. I haven't heard anything from @vrabaud or @yechun1 so I will be removing them as maintainers. Keeping this issue open until that task is complete.

cgdubois, 2021-11-26:
Hi @JWhitleyWork I am not sure what the normal procedure is for this, but I'd be interested in becoming a maintainer. I've been using ROS1 since 2017, although mostly on Kinetic but Melodic/Noetic aren't a big jump. I'm also interested in learning about ROS2 in the near term. Let me know if I can help/become involved.

JWhitleyWork, 2021-12-20:
@cgdubois Sorry for not responding sooner. Yes, I would be happy to have you help out with maintenance! You should have an invitation in your inbox.

closed 2024-05-30 by mikeferguson