ros-perception/image_pipeline #576: Meaning of distortion parameters values (default setting)

Harsharma2308, 2020-07-31:
What are the 5 distortion values I am getting on calibrating my camera using a pinhole model in the final yaml file?

My guess is [k1,k2,p1,p2,0] 
k1,k2- radial distortion co-effs
p1,p2- tangential distortion co-effs

Is there a documentation on it? Where can I read it?

trigal, 2020-07-31:
from the [wiki](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#double%20calibrateCamera(InputArrayOfArrays%20objectPoints,%20InputArrayOfArrays%20imagePoints,%20Size%20imageSize,%20InputOutputArray%20cameraMatrix,%20InputOutputArray%20distCoeffs,%20OutputArrayOfArrays%20rvecs,%20OutputArrayOfArrays%20tvecs,%20int%20flags,%20TermCriteria%20criteria))

distCoeffs – Output vector of distortion coefficients (k_1, k_2, p_1, p_2[, k_3[, k_4, k_5, k_6]]) of 4, 5, or 8 elements.

JWhitleyWork, 2020-07-31:
@Harsharma2308 Please re-open if you need more information.

Harsharma2308, 2020-07-31:
Thanks for the prompt reply. I dont know if I should create another issue just for this, but what is the default distortion model assumed? Is it plumb bob?

JWhitleyWork, 2020-07-31:
Everything I can find in the code and tests indicates that `plumb_bob` is the only one used here.

closed 2020-07-31 by JWhitleyWork