ros-perception/image_pipeline #589: Downsampling in calibrator.py makes detection fail

martimorta-at-work, 2020-08-21:
Hi,
I was having issues detecting a checkerboard until I changed the downsampling size (below) from VGA for the actual size of the camera output.
https://github.com/ros-perception/image_pipeline/blob/458ff618a53ec92713544c3ff24d2153975d8f3f/camera_calibration/src/camera_calibration/calibrator.py#L523

I think having that as a parameter would allow better calibrations unless I am missing something here.
I am happy to submit a PR with VGA size being a parameter if there aren't other implications.

Cheers!

JWhitleyWork, 2021-06-24:
@martimorta-at-work I know it's going on almost a year since you submitted this but we would be happy to accept a PR with this.

MedisettiRenukeswar, 2026-04-17:
Hi, I’d like to work on this issue.

My plan is to replace the hardcoded VGA resolution (640x480) in calibrator.py with configurable parameters, while keeping default values unchanged for backward compatibility.

Please let me know if this approach sounds good.