ros-perception/image_pipeline #603: extract_images node skips frames inconsistently

karsjen, 2020-09-24:
Using the raspicam node https://github.com/UbiquityRobotics/raspicam_node i shot multiple videos in 24 fps, saving the image/compressed topic to a rosbag. The extract_images node requires the user to enter the time between frames parameter. however, when this is set to 1/24 second it will extract approximately half of all frames. When set to zero the performance is better but it will still skip about 10 percent of all frames (comparing it to the sequence parameter in the header of the CompressedImage topic). There seems to be a random element involved as applying the same extraction node to the same rosbag multiple times will produce a slightly different amount of frames. Can an option be added which will save every frame instead of the user having to give the 'time between frames' parameter?

karsjen, 2020-09-24:
when playing the rosbag and starting the extract_images node from one launch file the performance is much better (missing about 2% of frames) and varying only by a few frames in number between applications on the same rosbag.

mikeferguson, 2024-01-19:
This is pretty much a duplicate of https://github.com/ros-perception/image_pipeline/issues/274 - the sec_per_frame basically only allows you to sample 1/2, 1/4, etc of the images coming in

closed 2024-01-19 by mikeferguson