ros-perception/image_pipeline #612: ros-noetic python interpretor has no 'python'

paszoujp, 2020-10-28:
I installed camera_calibration from apt-get
when I run cameracalibrator.py, this error occurs:
_/opt/ros/noetic/bin/rosrun: /opt/ros/noetic/lib/camera_calibration/cameracalibrator.py: /usr/bin/python: bad interpreter: No such file or directory
/opt/ros/noetic/bin/rosrun: line 150: /opt/ros/noetic/lib/camera_calibration/cameracalibrator.py: Success_

I then looked in /usr/bin/, then find that there's no file named python, only python2 or python3 something
I then edited cameracalibrator.py, changed the 1st line from `#!/usr/bin/python` to  `#!/usr/bin/python3`,then problem solved
Maybe you should update the python interpreter statement in each python file, for ubuntu 20.04 and ros noetic?

JWhitleyWork, 2020-12-12:
This should be fixed after 0a85550b361575478bf5ebf622d2f90d0a097795. Please pull the latest `noetic` branch and test with a source-build.

JWhitleyWork, 2021-01-25:
Closing due to lack of feedback.

closed 2021-01-25 by JWhitleyWork