ros-perception/image_pipeline #615: Conversion from point_cloud_xyz to depth image

AravindaDP, 2020-11-01:
Are there out of the box node/nodelets for conversion from point_cloud_xyz to depth_image?

Can we have a depth_image_proc/from_point_cloud_xyz node/nodelet for this purpose? If this is a welcome enhancement I can take a stab on on it and make a PR.

JWhitleyWork, 2020-12-11:
It is very rare that users want to go from a point cloud to a depth image. What is the use-case?

AravindaDP, 2020-12-12:
Yeah I guess you are right it's only for a niche of uses like converting regularly spaced point clouds (e.g. point clouds for 3D lidar like VLP-16) In my case it was for 3d point projection onto camera frame which used depth_image but later I came across a node that accepts point_cloud_xyz.
I also noted that in cases like 3d-lidar it also is bandwidth efficient to use depth_image instead point_cloud_xyz

JWhitleyWork, 2020-12-12:
Thanks for the explanation. Closing.

closed 2020-12-12 by JWhitleyWork