ros-perception/image_pipeline #618: Stereo calibration is failing at melodic

rahuldeo2047, 2020-11-21:
I see one fix thought, the issue 503. Did not work either.

reference: 
https://github.com/ros-perception/image_pipeline/pull/542/commits/d0bf180999d1f56c254de14d2e2df7145e8c2b0e
https://github.com/ros-perception/image_pipeline/issues/503

I cloned image_pipeline melodic, built, and used.
`$ rosrun camera_calibration cameracalibrator.py --approximate 0.1 --size 9x6 --square 0.15 right:=/stereo/right/image_raw left:=/stereo/left/image_raw right_camera:=/stereo/right left_camera:=/stereo/left
`
I also tried 
`$  ./src/image_pipeline/camera_calibration/nodes/cameracalibrator.py --approximate 0.1 --size 9x6 --square 0.15 right:=/stereo/right/image_raw left:=/stereo/left/image_raw right_camera:=/stereo/right left_camera:=/stereo/left
`
Same error log

```
**** Calibrating ****
mono fisheye calibration...
mono fisheye calibration...
stereo fisheye calibration...
OpenCV Error: Assertion failed (svd.w.at<double>(0) / svd.w.at<double>((int)svd.w.total() - 1) < thresh_cond) in CalibrateExtrinsics, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/calib3d/src/fisheye.cpp, line 1427
Traceback (most recent call last):
  File "/home/rahul/Work/ROS/camera/src/image_pipeline/camera_calibration/src/camera_calibration/camera_calibrator.py", line 269, in on_mouse
    self.c.do_calibration()
  File "/home/rahul/Work/ROS/camera/src/image_pipeline/camera_calibration/src/camera_calibration/calibrator.py", line 1401, in do_calibration
    self.cal_fromcorners(self.good_corners)
  File "/home/rahul/Work/ROS/camera/src/image_pipeline/camera_calibration/src/camera_calibration/calibrator.py", line 1173, in cal_fromcorners
    flags = flags)
cv2.error: /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/calib3d/src/fisheye.cpp:1427: error: (-215) svd.w.at<double>(0) / svd.w.at<double>((int)svd.w.total() - 1) < thresh_cond in function CalibrateExtrinsics
```
Checked for monocular camera pinhole and stereo camera pinhole working great. Using the original ros-melodic installation.

JWhitleyWork, 2020-12-11:
@rahuldeo2047 Please provide the following information:

- Operating System
- ROS Version
- How ROS was Installed (built from source or from official repositories)
- Which commit you are using from this repository
- Which version of OpenCV you are using

You also say that you cloned from source and built the package but you also say "Using the original `ros-melodic` installation." - what does this mean?

rahuldeo2047, 2020-12-16:
Two ways I tried 

**Way 1**
The official ros install also is creating this issue
Operating System: Ubuntu 18.04
How ROS was Installed (built from source or from official repositories): official
Which commit you are using from this repository: official
Which version of OpenCV you are using: the OpenCV comes with ros melodic, 3.2.0+dfsg-4ubuntu0.1

**Way 2** 
Then while googling I came across issue #503 . Accordingly, I cloned the image pipeline and did catkin_make.
And ran. Same issue.
Operating System: Ubuntu 18.04
How ROS was Installed (built from source or from official repositories): ros official
**Which commit you are using from this repository**: https://github.com/ros-perception/image_pipeline/tree/melodic latest
Which version of OpenCV you are using: the OpenCV comes with ros melodic, 3.2.0+dfsg-4ubuntu0.1

Thank you for the question for clarity.

rahuldeo2047, 2020-12-16:
Using the original ros-melodic installation
means 

fisheye calibration is failing from stereo in melodic and also with image_pipeline package
the pinhole camera calibration is working with melodic.

original ros-melodic
not using the cloned image_pipeline package 
using the officially installed image_pipeline package.

KUTuaNithid, 2021-01-14:
Did you try with a lower resolution?

mikeferguson, 2024-01-17:
Melodic is now EOL

closed 2024-01-17 by mikeferguson