ros-perception/image_pipeline #619: How to publish rtsp stream as ros2 topic using image_publisher?

Kaju-Bubanja, 2020-11-30:
According to https://discourse.ros.org/t/ros-2-vid... I should be able to just call `ros2 run image_publisher image_publisher_node rtsp://192.168.254.151/live_stream` and the rtsp stream should be published as a ros2 topic. I can view the rtsp stream using vlc just fine, but when trying above command I get following error after a while:
```

[h264 @ 0x5639cdf9f600] mb_type 58 in P slice too large at 81 36 
[h264 @ 0x5639cdf9f600] error while decoding MB 81 36
```

and sometimes the error is:

```
[h264 @ 0x5651197324c0] cbp too large (111) at 1 59
[h264 @ 0x5651197324c0] error while decoding MB 1 59
```

I built image_publisher and cv_bridge from source using the ros2 branch and I tried with the default opencv version 4.2.0 and now also with source built opencv version 4.5.1 hoping that it might be fixed in a newer version, but I still receive the same error. Any idea how I can fix that error?

Kaju-Bubanja, 2020-11-30:
Even though I restarted the camera already once, I restarted it again and now it works. I have no clue why it didn't work and why it works now. Computers are black magic. If somebody knows more I'm glad to hear about it.

closed 2020-11-30 by Kaju-Bubanja

JWhitleyWork, 2020-11-30:
Glad you got it resolved!

Kaju-Bubanja, 2021-01-06:
Actually I later found out, that this was happening to network congestion. If there is not enough throughput you might see this warning. When increasing the network throughput I got these message less often, but still they popped up from time to time and in the video feed I saw some artifacts. As such I think this error message should maybe be more of a warning than an error.