ros-perception/image_pipeline #621: sync with Noetic

dmorris0, 2020-12-11:
There was an important bug fix that is needed to run camera calibration in Noetic: pull #552
However, this is not in the current Noetic release, and it looks like the most recent sync with Noetic in October did not include ros-perception. Can we get a sync and new release soon?

(Not sure if this is the right place to be posting this).

Thanks

maximus5684, 2020-12-11:
Thanks for the reminder! The PR is in: https://github.com/ros/rosdistro/pull/27751

However, they are currently in a package-sync freeze and it may not make it in this time. If it does not make it in, you can build from source until the next package sync.

JWhitleyWork, 2020-12-11:
Whoops - wrong account.

dmorris0, 2020-12-11:
Ah, okay.  Yes, I built it from source, and that works well.  Thanks!

closed 2020-12-11 by dmorris0