ros-perception/image_pipeline #622: [Realsense T265] stereo_camera_model.cpp:35 issue

srnetsr, 2020-12-12:
Hi.
I am running T265 with the following launch file.

<launch>
    <remap from="/stereo/left/image_raw" to="/camera/infra1/image_rect_raw"/>
    <remap from="/stereo/left/camera_info" to="/camera/infra1/camera_info"/>
    <remap from="/stereo/right/image_raw" to="/camera/infra2/image_rect_raw"/>
    <remap from="/stereo/right/camera_info" to="/camera/infra2/camera_info"/>
    
    <group ns='stereo'>
        <node name="stereo_image_proc" pkg="stereo_image_proc" type="stereo_image_proc" output='screen'>
        <param name="approximate_sync" value="True" />
        </node>
    </group>
</launch>

and I am trying to follow [stereo_image_proc](http://wiki.ros.org/stereo_image_proc?distro=noetic)
But I got the following error with `$ rosrun image_view stereo_view stereo:=/stereo image:=image_rect`

**/image_geometry/src/stereo_camera_model.cpp:35: bool image_geometry::StereoCameraModel::fromCameraInfo(const CameraInfo&, const CameraInfo&): Assertion `left_.tfFrame() == right_.tfFrame()' failed.**

Is there anyone who know the solution for this issue?

JWhitleyWork, 2020-12-14:
This looks like it's saying that the `frame_id` field in the header of the images don't match. They must be the same.

srnetsr, 2020-12-14:
@JWhitleyWork  Thank you for your reply.
I confirmed that T265's fisheye camera's frame_id is `camera_fisheye1_optical_frame`

What the header of the images should suppose to be?

srnetsr, 2020-12-16:
@JWhitleyWork 
here is update.

I make the same `frame_id` of T265's two fisheye cameras as `camera_fisheye_optical_frame`
So when I run stere_image_proc

```
[ WARN] [1608161695.586726659]: Color topic '/stereo/left/image_color' requested, but raw image data from topic '/camera/fisheye1/image_raw' is grayscale
terminate called after throwing an instance of 'image_geometry::Exception'
terminate called recursively
  what():  Cannot call rectifyImage when distortion is unknown.
```

Could you please help me to resolve this issue?
I am using T265 camera now.

mikeferguson, 2024-01-17:
Configuration questions like this are generally better for https://robotics.stackexchange.com

closed 2024-01-17 by mikeferguson