ros-perception/image_pipeline #625: Topic /points cant show in rviz

Schloern93, 2021-01-12:
Hallo,

this is probably a rviz specific question.
I use the package "depth_image_proc/point_cloud_xyz".
With ros2 topic echo I can view my PointCloud2 message (Topic: "/points"). 
When I try to display the PointCloud2 message in rviz, rviz does not show any points. 
But the topic /points is found and I can view it with ros2 topic echo /points.
Thanks for the help.

Schloern93, 2021-01-12:
I solved it by changing this line:
`pub_point_cloud_ = create_publisher<PointCloud2>("points", rclcpp::SensorDataQoS());`
to
`pub_point_cloud_ = create_publisher<PointCloud2>("points", 1);`

Can someone explain to me the reason for this?

swiz23, 2021-09-10:
I am running into the same issue as well - though I didn't try to modify the code.

JWhitleyWork, 2021-09-13:
Once you subscribe to the topic in `rviz2`, you can expand the topic details and change the QoS settings to match the `SensorDataQoS` settings. Here they are: https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1SensorDataQoS.html. If this works for you, please let me know and I will close this issue.

swiz23, 2021-09-13:
Thanks @JWhitleyWork, that did the trick!!

I specifically had to change the PointCloud2 -> Topic -> Reliability Policy in Rviz2 from 'Reliable' to 'Best Effort' for the pointcloud to show.

JWhitleyWork, 2021-09-13:
Excellent! Closing.

closed 2021-09-13 by JWhitleyWork