ros-perception/image_pipeline #631: Unable to run camera_calibration cameracalibration.py

zactodd, 2021-02-04:
Hello,

I was running using melodic and using my laptop camera.

I ran this
```
rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 image:=/cv_camera/image_raw camera:=/cv_camera
```

These are the available topic
```
/cv_camera/camera_info
/cv_camera/image_raw
/cv_camera/image_raw/compressed
/cv_camera/image_raw/compressed/parameter_descriptions
/cv_camera/image_raw/compressed/parameter_updates
/cv_camera/image_raw/compressedDepth
/cv_camera/image_raw/compressedDepth/parameter_descriptions
/cv_camera/image_raw/compressedDepth/parameter_updates
/cv_camera/image_raw/theora
/cv_camera/image_raw/theora/parameter_descriptions
/cv_camera/image_raw/theora/parameter_updates
```

this is the error msg
```

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/camera_calibrator.py", line 108, in run
    self.function(m)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/camera_calibrator.py", line 189, in handle_monocular
    drawable = self.c.handle_msg(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/calibrator.py", line 811, in handle_msg
    gray = self.mkgray(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/calibrator.py", line 280, in mkgray
    if self.br.encoding_to_dtype_with_channels(msg.encoding)[0] in ['uint16', 'int16']:
  File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)

(display:24905): GLib-GObject-CRITICAL **: 16:57:40.413: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(display:24905): GLib-GObject-CRITICAL **: 16:57:40.517: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
```

Any help would be great.

Cheers,
Zac

yuma116, 2021-02-06:
I recommend checking the issue: https://github.com/ros-perception/image_pipeline/issues/201

mikeferguson, 2024-01-17:
Melodic is EOL

closed 2024-01-17 by mikeferguson