ros-perception/image_pipeline #637: ros-melodic-fisheye-calibration-solved

jwrl7, 2021-02-19:
Hey folks,

Wanted to share some recent findings and solution for calibrating fisheye cameras FOV132 and large image size 1080p using the latest melodic branch and OpenCV3.4.4. 

Solution: Modify the downsample_and_detect(self, img) function
```

  #calibrator.py line 515    
  def downsample_and_detect(self, img):
        """
        Downsample the input image to approximately VGA resolution and detect the
        calibration target corners in the full-size image.

        Combines these apparently orthogonal duties as an optimization. Checkerboard
        detection is too expensive on large images, so it's better to do detection on
        the smaller display image and scale the corners back up to the correct size.

        Returns (scrib, corners, downsampled_corners, ids, board, (x_scale, y_scale)).
        """
        # Scale the input image down to ~VGA size
      
        height = img.shape[0]
        width = img.shape[1]
        scale = math.sqrt( (width*height) / (640.*480.) )

        #NO... don't do this...
        """
        if scale > 1.0:
            scrib = cv2.resize(img, (int(width / scale), int(height / scale)))
        else:
            scrib = img
        # Due to rounding, actual horizontal/vertical scaling may differ slightly
        x_scale = float(width) / scrib.shape[1]
        y_scale = float(height) / scrib.shape[0]
        """
        # Do this
        scrib = img
        x_scale = 1.
        y_scale = 1.

```

I was experiencing a handful of similar issues that many where reporting:
Distortion parameters returned are all zero or, 
when they were not zeros, the K matrix had a skew parameter, aka element (0,1) was non zero. 
Over several attempts of calibrating I had one of the above outcomes, even while passing the below params in various combinations.
```
--fisheye-fix-skew
-k 4
--fisheye-k-coefficients=4
```
The fix for me was commenting out code that down scales the image be for computing the checkerboard pattern. The reason being was going from 1080p to VGA (hard coded in calibrator.py) has a lot of room for rounding errors when the checkerboard corners are scaled back up to the original size. 

So i commented out the downs scaling, which is there because, yes it may take longer to find the checkerboard pattern, but only IF the checkerboard takes up majority of the image.

JWhitleyWork, 2021-02-26:
@jwrl7 Which version of ROS are you using? Did you build the repository from source or install from a repository? Which operating system? Which version of the package? What branch (if built from source)?

Are you willing to create a pull request to make the downsampling optional?

jwrl7, 2021-02-26:
Config:
Ubuntu 18.04
ROS melodic
Opencv 3.4.4 with cuda built from source
image_pipeline (melodic latest pull) built against Opencv 3.4.4 with cuda

I can put a pull request together but wont be immediate.

JWhitleyWork, 2021-12-06:
Closing for now. Please create a pull request if possible.

closed 2021-12-06 by JWhitleyWork

Snowbound9173, 2023-10-04:
This issue is still current, I can't even tell you how much time I spent looking for why I couldn't be more than about 2.5m away from my 4K camera with 120degree lens. Well I suspected downscaling since the visual image showed such poor resolution but I'm very very glad to have found this from @jwrl7 . Thank you very much I was able to mash your code change in and now I'm able to get that checkerboard found from 5-10 meters away thankfully! Thank you.

JWhitleyWork, 2023-10-04:
@Snowbound9173 Would you be willing to create a PR to integrate these changes? I'd be happy to review it, I just don't have time to create it at the moment.