ros-perception/image_pipeline #639: camera calibration error using image_pipeline

paul3333, 2021-02-26:
I am running your package on ubuntu 16.04, ROS Kinetic and been able to build all packages using catkin_make without errors.
I also included the image_pipeline package as you were using it to rectify the drones images.

When running the camera calibration
rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 image:=/ardrone/front/image_raw camera:=/ardrone/front/

I receive the following error:

`kinetic@kinetic:~$ rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 image:=/ardrone/front/image_raw camera:=/ardrone/front/
('Waiting for service', '/ardrone/front/set_camera_info', '...')
OK
Exception in thread Thread-4:
Traceback (most recent call last):
File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
self.run()
File "/home/kinetic/catkin_ws/src/image_pipeline/camera_calibration/src/camera_calibration/camera_calibrator.py", line 107, in run
self.function(m)
File "/home/kinetic/catkin_ws/src/image_pipeline/camera_calibration/src/camera_calibration/camera_calibrator.py", line 181, in handle_monocular
max_chessboard_speed = self._max_chessboard_speed)
File "/home/kinetic/catkin_ws/src/image_pipeline/camera_calibration/src/camera_calibration/calibrator.py", line 725, in init
super(MonoCalibrator, self).init(*args, **kwargs)
TypeError: super() argument 1 must be type, not classobject`

Has anyone faced this issue? Any help would be highly appreciated.

JWhitleyWork, 2021-02-26:
@paul3333 Did you build image_pipeline from source or install it from the repository? Which version are you using? If installed from source, which branch? Which operating system?

paul3333, 2021-02-26:
I have built image_pipeline by git cloneing from this image_pipeline github website. I installed the kinetic branch but received the same error also when installing the melodic branch.

My operating system is Ubuntu 16.04

Thanks for your help.

anjulo, 2022-05-13:
This helped me solve the problem.
https://stackoverflow.com/questions/1713038/super-fails-with-error-typeerror-argument-1-must-be-type-not-classobj-when

JWhitleyWork, 2022-07-07:
Hopefully the above comment helps. Additionally, Kinetic is no longer supported so I can't help much.