ros-perception/image_pipeline #643: False Depth Image from depth_image_pro/register

Marki2112, 2021-03-19:
Hello everyone,

i using the realsense d435 and ROS Melodic to transform the depth image from "camera_depth_optical_link" to the "camera_color_optical_link". I am using a rosbag to get the data, but the result look like for the transformed depth image not correct. My current launch-file can you see here
`<launch>
    
    <arg name="color_camera_info" default="/camera/color/camera_info"/>
    <arg name="depth_camera_info" default="/camera/depth/camera_info"/>
    <arg name="depth_image" default="/camera/depth/image_rect_raw"/>
    
    <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager"/>

    <node pkg="nodelet" type="nodelet" name="depth2rgbFrame"
          args="load depth_image_proc/register nodelet_manager">
      
      
      <remap from="rgb/camera_info" to="$(arg color_camera_info)"/>
      <remap from="depth/camera_info" to="$(arg depth_camera_info)"/>
      <remap from="depth/image_rect" to="$(arg depth_image)"/>

    </node>
</launch>`

Here you can see the difference between the normal depth image  and the transformed depth image:
normal depth image 
[image]
[image]

lucasw, 2022-01-16:
Try setting `fill_upsampling_holes` to true.

mikeferguson, 2024-01-19:
There is a suggested fix, and this sort of thing should probably go to https://robotics.stackexchange.com in the future

closed 2024-01-19 by mikeferguson