ros-perception/image_pipeline #645: unable to build package

fhlb-tech, 2021-03-27:
when trying to build using:
catkin build -j4

I get the following output:
Errors     << depth_image_proc:make /home/robot/catkin_ws/logs/depth_image_proc/build.make.016.log                                                                     
/home/robot/catkin_ws/src/Dependencies/image_pipeline/depth_image_proc/src/nodelets/point_cloud_xyz_radial.cpp: In function ‘cv::Mat depth_image_proc::initMatrix(cv::Mat, cv::Mat, int, int, bool)’:
/home/robot/catkin_ws/src/Dependencies/image_pipeline/depth_image_proc/src/nodelets/point_cloud_xyz_radial.cpp:104:6: error: ‘undistortPoints’ is not a member of ‘cv’
  cv::undistortPoints(sensorPoints, undistortedSensorPoints, cameraMatrix, distCoeffs);
      ^~~~~~~~~~~~~~~
/home/robot/catkin_ws/src/Dependencies/image_pipeline/depth_image_proc/src/nodelets/point_cloud_xyzi_radial.cpp: In member function ‘cv::Mat depth_image_proc::PointCloudXyziRadialNodelet::initMatrix(cv::Mat, cv::Mat, int, int, bool)’:
/home/robot/catkin_ws/src/Dependencies/image_pipeline/depth_image_proc/src/nodelets/point_cloud_xyzi_radial.cpp:124:6: error: ‘undistortPoints’ is not a member of ‘cv’
  cv::undistortPoints(sensorPoints, undistortedSensorPoints, cameraMatrix, distCoeffs);
      ^~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/depth_image_proc.dir/src/nodelets/point_cloud_xyz_radial.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/depth_image_proc.dir/src/nodelets/point_cloud_xyzi_radial.cpp.o] Error 1
make[1]: *** [CMakeFiles/depth_image_proc.dir/all] Error 2
make: *** [all] Error 2
cd /home/robot/catkin_ws/build/depth_image_proc; catkin build --get-env depth_image_proc | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -

I am on melodic with openCV4 (contrib with non-free modules) built and installed from source. I am building my workspace with python3 (cv_bridge also installed from source).  I even tried:
catkin build -j4 -DOpenCV_DIR=/usr/local/lib/cmake/opencv4
the same error message appeared. 

I tried installing cv_bridge and openCV3 from apt-get but nothing changed

JWhitleyWork, 2021-03-28:
When you say you are "on melodic" does that mean that you are also using the `melodic` branch of this repo? Additionally, `melodic` is not designed to be used with OpenCV4 because it is not the default that comes with Ubuntu 18.04. See [REP-3](https://ros.org/reps/rep-0003.html) for the list of platform support requirements. Additionally, there is no guarantee that these packages will build at all with a custom OpenCV installation.

fhlb-tech, 2021-03-30:
Hello and Thank you for your response.
I am using the melodic branch.
I also tried using the neotic branch (compiling it with python3) and got the same error.
I have a manually compiled and installed openCV4 with the ROS package cv_bridge (compiled with python3) built from source. I did that because I need some features of openCV4.
I can use apt-get to install the image_pipeline packages. This will install opencv3 libs and cv_bridge. Will this cause any probelms as I have now two installed openCV and two installed ROS packages cv_bridge. How will a ROS node choose where to find its dependencies (whether to look for ROS packages installed from apt-get or locally built packages). Everything seems to be working but I am afraid to break other packages relying on cv_bridge and openCV4

redchillipadi, 2021-07-23:
I had the same issue using Focal fossa with ros-noetic and opencv 4.2.0.

The undistortPoints function is in opencv2/calib3d.hpp and I was able to fix the issue by adding
#include <opencv2/calib3d.hpp>
to depth_image_proc/src/nodelets/point_cloud_xyz_radial.cpp
and depth_image_proc/src/nodelets/point_cloud_xyzi_radial.cpp
and depth_image_proc/src/nodelets/point_cloud_xyzrgb_radial.cpp

The same fix was also required for StereoBM as I reported in issue #685

moksh019, 2022-01-06:
Thanks  @redchillipadi , it worked and solved the problem. 
I am wondering that I had built this repo long time ago with no such error. But I could not understand why it was generating the error this time for me !

closed 2022-01-06 by JWhitleyWork