ros-perception/image_pipeline #655: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

abdul-mannan-khan, 2021-05-07:
Before I start any further discussion, please note that I have googled this error in general and tried to resolve the issue. I could not succeed. So, here is what is going on. 

**How can I solve the problem of using GTK+ 2.x and GTK+3 in the same process for image_view package?** 
 
When I try to run image_view package, I get this error. 
```
rosrun image_view image_view image:=/raspicam_node/image_raw
[ INFO] [1620424536.219092141]: Initializing nodelet with 4 worker threads.
[ INFO] [1620424536.313048155]: Using transport "raw"

(image_raw:1672): Gtk-ERROR **: 06:55:36.317: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
Trace/breakpoint trap
```
To find the solution, I have gone through many blogs such as [https://stackoverflow.com/questions/42534129/opencv-error-gtk-2-x-symbols-detected-using-gtk-2-x-and-gtk-3-in-the-same-p](here), [https://askubuntu.com/questions/234379/what-does-gtk-error-gtk-2-x-symbols-detected-mean](here) and [https://stackoverflow.com/questions/20714343/gtk-error-gtk-2-x-symbols-detected](here) and so on. One thing which is clear to me is that image_view package is using GTK+ 2.x and GTK+ 3 in the same process which is not supported. Now, I have tried going through CMakeLists.txt of image_view package. I found this section of the code at line 41-43.  

```
find_package(GTK2)
add_definitions(-DHAVE_GTK)
include_directories(${GTK2_INCLUDE_DIRS})
```

I have both GTK2 and GTK3 as shown here by the command 
`pi@raspberrypi:~/ros1_ws $ dpkg -l libgtk* | grep -e '^i' | grep -e 'libgtk-*[0-9]'`
which game the following output. 
```
ii  libgtk-3-0:armhf             3.24.5-1+rpt2  armhf        GTK+ graphical user interface library
ii  libgtk-3-common              3.24.5-1+rpt2  all          common files for the GTK+ graphical user interface library
ii  libgtk2.0-0:armhf            2.24.32-3+rpt1 armhf        GTK+ graphical user interface library
ii  libgtk2.0-bin                2.24.32-3+rpt1 armhf        programs for the GTK+ graphical user interface library
ii  libgtk2.0-common             2.24.32-3+rpt1 all          common files for the GTK+ graphical user interface library
ii  libgtk2.0-dev:armhf          2.24.32-3+rpt1 armhf        development files for the GTK+ library
```

So, I suspect that image_view should work fine  but I do not know. So, my question is how can I run image_view package without any hassle of GTK+ 2.x and GTK+3 errors? Should I change my CMakeLists.txt or something else. Any help would be appreciated. Thank you.

JWhitleyWork, 2021-05-08:
@abdul-mannan-khan Which commit/branch/version of `image_view` are you using? Did you build from source or install from a package? What operating system and version?

abdul-mannan-khan, 2021-05-09:
Dear @JWhitleyWork, 

Thank you so much for your attention. 

1. I am using it on Raspberry Pi 4. 
2. I am using their recommended software "Raspberry Pi OS with desktop and recommended software" given in official webpage [here](https://www.raspberrypi.org/software/operating-systems/#raspberry-pi-os-32-bit)
3. I am using ROS Melodic. 
4. I installed ROS Melodic using the method described in ROS official web-page as [here](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Melodic%20on%20the%20Raspberry%20Pi). It builds it from the source, I guess. I am not sure though. Sorry for being not sure. 
5.  I am using melodic branch of this `image_view`. 

I am really grateful to you for your kind favor.

mikeferguson, 2024-01-20:
Melodic is EOL