ros-perception/image_pipeline #656: image_view and camera_calibration not working for gazebo camera

Kaiser-Wang, 2021-05-12:
Hi, I'm using image_view and camera_calibration to preview and calibrate my webcam and they worked really well.
However, when I used them for a simulated gazebo camera with a published topic /camera/image_raw, it didn't work.
I used the following command to view /camera/image_raw topic:
`ros2 run image_view image_view --ros-args --remap image:=/camera/image_raw`
and it printed the following line but didn't open up the preview window:
`[INFO] [1620803840.239610932] [image_view_node]: Using transport "raw"`
Same thing happened for the camera calibration:
`ros2 run camera_calibration cameracalibrator --size 9x6 --square 0.025 --ros-args image:=/camera/image_raw`
which returns
`Waiting for service camera/set_camera_info ...`
`OK`
`Waiting for service left_camera/set_camera_info ...`
`OK`
`Waiting for service right_camera/set_camera_info ...`
`OK`
Again, it didn't open up the calibration window.
I tested this with my webcam and it worked properly.
I'm using Ubuntu 20.04 and ROS2 foxy. Thank you.

bastianhjaeger, 2021-09-20:
Facing a similar behavior. Did you resolve your problem?

mikeferguson, 2024-01-17:
Looks like this is a QoS mismatch - we're going to clobber these in https://github.com/ros-perception/image_pipeline/issues/847

closed 2024-01-17 by mikeferguson