ros-perception/image_pipeline #658: Camera Calibrator - Noetic - /usr/bin/python: bad interpreter

Teo-Cardoso, 2021-05-19:
I've been facing a problem running the camera calibrator in ROS Noetic ( Ubuntu 20.04 ).

running this rosrun example:
`rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 image:=/my_camera/image camera:=/my_camera`

the output:
`/opt/ros/noetic/bin/rosrun: /opt/ros/noetic/lib/camera_calibration/cameracalibrator.py: /usr/bin/python: bad interpreter: No such file or directory`

This problem can be easily solved by changing the first line from:
`#!/usr/bin/python`
to 
`#!/usr/bin/env python3`

Is it my error or should be replaced in the repository?

nstolyarov, 2021-05-28:
I have same the same error to. I guess this happens due to Noetic supports python3 packages instead of python2.

christian-rauch, 2021-07-09:
I am getting the same error and explicitly requesting `python3` is the correct thing to do. @mjcarroll Can you have a look at this?

FSund, 2021-09-10:
This is fixed on the `noetic` branch via 0a85550, but the shebang is still `#!/usr/bin/python` on my installation. I have the following version installed: `ros-noetic-camera-calibration-parsers/focal,now 1.12.0-1focal.20210727.070336 amd64`

christian-rauch, 2021-09-10:
> but the shebang is still `#!/usr/bin/python` on my installation

The patch you linked was committed in Dec 2020, after the last release of `image_pipeline` (version 2.1.1 in Jun 2020).

But even then, `#!/usr/bin/env python` will give you the python2 interpreter by default on Ubuntu. You either have to install the `python-is-python3` package, which links `python` to `python3`, or call that script directly with python3.

FSund, 2021-09-10:
> > but the shebang is still `#!/usr/bin/python` on my installation
> 
> The patch you linked was committed in Dec 2020, after the last release of `image_pipeline` (version 2.1.1 in Jun 2020).

Thanks, I was wondering why it was missing on my install.

> But even then, `#!/usr/bin/env python` will give you the python2 interpreter by default on Ubuntu. You either have to install the `python-is-python3` package, which links `python` to `python3`, or call that script directly with python3.

I'm using Pop!OS 20.04, where `python` is not linked to anything by default. I'm not sure what the case is on Ubuntu 20.04. When I clone the latest `noetic` and compile via `catkin build`, running

    rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 image:=/my_camera/image camera:=/my_camera

works without errors.

According to [this](http://wiki.ros.org/UsingPython3/SourceCodeChanges#Changing_shebangs) article, `#!/usr/bin/env python` is the correct shebang, because it gets converted to the correct one via `catkin_install_python()`. 

This seems to be working as intended on my machine:

```
filip@pop-os:~/ros/image_pipeline_ws$ cat devel/lib/camera_calibration/cameracalibrator.py 
#!/usr/bin/python3
# -*- coding: utf-8 -*-
# generated from catkin/cmake/template/script.py.in
# creates a relay to a python script source file, acting as that file.
# The purpose is that of a symlink
python_script = '/home/filip/ros/image_pipeline_ws/src/image_pipeline/camera_calibration/nodes/cameracalibrator.py'
with open(python_script, 'r') as fh:
    context = {
        '__builtins__': __builtins__,
        '__doc__': None,
        '__file__': python_script,
        '__name__': __name__,
        '__package__': None,
    }
    exec(compile(fh.read(), python_script, 'exec'), context)
```

JWhitleyWork, 2021-12-06:
This just requires a new release of the `noetic` branch to the build farm which has already been requested in #527. Please build from source until that ticket is resolved. If you encounter a different error when building from source, please create a new issue.

closed 2021-12-06 by JWhitleyWork