ros-perception/image_pipeline #668: `depth_image_proc/point_cloud_xyzrgb` doesn't publish point cloud when starting with other nodes in a single launch file

lianghongzhuo, 2021-06-16:
`depth_image_proc/point_cloud_xyzrgb` doesn't publish point cloud when starting with other nodes in a single launch file.
When I launch kinect2_bridge, for example, https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/launch/kinect2_bridge.launch#L84-L112
The above launch file will publish depth, RGB, in a given publish rate, however, the point cloud pressing node doesn't always publish point clouds. Sometimes it works, sometimes it doesn't publish the desired point cloud. Restart the launch file above several times can help, but not always work.
A workaround for me is to cut these code in another launch file and launch it indviduly. So I want to ask if this issue only happens to me or is common to other people. I tried both image_pipline come with the Debian and the source install.
Other configurations are ubuntu 18.04 and ROS melodic.

lianghongzhuo, 2021-06-16:
I admit this is hard to debug, even I can not reproduce it at the moment.

lianghongzhuo, 2021-06-16:
seems not the image_pipine issue, the problem might be the multiple nodelet in the same launch file. Put the depth to point cloud part node to another launch file and include it in the `kinect2_bridage.launch` also works.

mikeferguson, 2024-01-18:
I have not seen this in either ROS 1 or ROS 2 - and I use xyzrgb quite extensively. This ticket was also filed against 18.04/Melodic, which is now end of life. If it re-occurs on a supported distro, perhaps come up with a bagfile + launch file to reproduce and file a new reopen ticket with that detail.

closed 2024-01-18 by mikeferguson