ros-perception/image_pipeline #682: fisheye calibration file of the image_pipleline calibration node doesn't use correct model for rectification.

bjajoh, 2021-07-16:
Hi,

I used the integrated camera calibration tool with the fisheye model.
The camera is been calibrated nicely and the calibration tool shows a good rectified image.
However when the exported ost.yaml is loaded in the "camera_info_url" and processed with image_proc, the rectification is not working correctly. I assume the distrortion model is wrong, however I'm not sure how to solve it or if it is an actual bug.
I'm using Melodic.
Here my ost.yaml calibration file:
`image_width: 1640
image_height: 1232
camera_name: csi_cam_0
camera_matrix:
  rows: 3
  cols: 3
  data: [ 403.97646,   -0.96635,  871.2187 ,
            0.     ,  404.70652,  597.37327,
            0.     ,    0.     ,    1.     ]
camera_model: fisheye
distortion_coefficients:
  rows: 1
  cols: 4
  data: [0.175653, -0.031772, 0.000362, -0.000719]
rectification_matrix:
  rows: 3
  cols: 3
  data: [ 1.,  0.,  0.,
          0.,  1.,  0.,
          0.,  0.,  1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [ 403.97646,   -0.96635,  871.2187 ,    0.     ,
            0.     ,  404.70652,  597.37327,    0.     ,
            0.     ,    0.     ,    1.     ,    0.     ]
`

frequiem11, 2021-08-15:
Any update on the issue? I'm in the same boat.

bjajoh, 2021-08-16:
Here is my working config:

`image_width: 1640
image_height: 1232
camera_name: csi_cam_0
camera_matrix:
  rows: 3
  cols: 3
  data: [ 403.97646,   -0.96635,  871.2187 ,
            0.     ,  404.70652,  597.37327,
            0.     ,    0.     ,    1.     ]
#camera_model: fisheye
distortion_model: equidistant
distortion_coefficients:
  rows: 1
  cols: 4
  data: [0.175653, -0.031772, 0.000362, -0.000719]
rectification_matrix:
  rows: 3
  cols: 3
  data: [ 1.,  0.,  0.,
          0.,  1.,  0.,
          0.,  0.,  1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [ 403.97646,   -0.96635,  871.2187 ,    0.     ,
            0.     ,  404.70652,  597.37327,    0.     ,
            0.     ,    0.     ,    1.     ,    0.     ]`

You need to replace `#camera_model: fisheye` with `distortion_model: equidistant`.
This is definitely a bug/incompatibility!

JWhitleyWork, 2021-08-16:
@bjajoh Would you mind submitting a pull request to fix the issue?

bjajoh, 2021-08-17:
@JWhitleyWork Yes, I'll do it soon :)

chinitaberrio, 2021-10-13:
@bjajoh How did you run the node for fisheye calibration? I'm getting only zeros for the distortion coefficients :(

tik0, 2021-11-24:
Hi @bjajoh , what do you mean with `rectification is not working correctly`? I came to this issue, because the image rectification using image_proc does not work under Melodic for me at all. If one looks into `image_geometry`, neither `fisheye` nor `equidistant` models are processed. Only neotic is able to handle the rectification of `equidistant`. Under melodic it gives me:

```
$ ROS_NAMESPACE=camera rosrun image_proc image_proc
terminate called after throwing an instance of 'image_geometry::Exception'
  what():  Cannot call rectifyImage when distortion is unknown.
Aborted (core dumped)
```

cgdubois, 2021-11-25:
I was having this problem too, and noticed in https://github.com/ros-perception/image_pipeline/blob/noetic/camera_calibration/src/camera_calibration/calibrator.py on line 330 it appears to be hard coded to use CAMERA_MODEL.PINHOLE. I think set_camemode is supposed to change self.camera_model?

I hardcoded it to CAMERA_MODEL.FISHEYE and I think I got the calibration data I needed. I'm just verifying with others but can make a pull request to hopefully fix this if someone doesn't beat me to it.

arjunskumar, 2022-03-11:
@chinitaberrio had you able to run it?

IamPhytan, 2022-03-30:
Similar issues on ros melodic too
I will try it in a VM with ros noetic

Runepx4, 2022-05-05:
I got it working using ros noetic. Eg.: `ROS_NAMESPACE=camera/fisheye1/ rosrun image_proc image_proc`

mikeferguson, 2024-02-01:
It would appear that this was an issue with melodic due to an older image_geometry package - melodic is now EOL, and it looks like this is working in both Noetic and ROS2

closed 2024-02-01 by mikeferguson