ros-perception/image_pipeline #684: Waiting for service /usb_cam/set_camera_info ... Service not found QMutex: destroying locked mutex Segmentation fault (core dumped)

virgin-coder, 2021-07-17:
I am using ros neotic version. I followed the steps from [https://github.com/mattsinbot/Pose-Estimation-Aruco-Marker-Ros](url), I am running the command " rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.108 image:=/usb_cam/image camera:=/usb_cam"
And I get above error, any idea why it is so ?

virgin-coder, 2021-07-17:
Working on ubuntu 20.04.

virgin-coder, 2021-07-19:
Done, needed to run usb_camera_node via rosrun usb_cam usb_cam_node before doing it.

closed 2021-07-19 by virgin-coder

AlexWUrobot, 2022-11-21:
I also try "rosrun usb_cam usb_cam_node" and "roscore" in two separate terminals.

In the third terminal, 
I have the similar issue when running "rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.0245 image:=/camera/image_raw camera:=/camera" 
I simply remove "camera:=/camera" and just run "rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.0245 image:=/camera/image_raw"

JWhitleyWork, 2022-11-21:
@AlexWUrobot If you are having a similar issue, please do not bump zombie issues - create a new one and provide the information requested in the issue template.