ros-perception/image_pipeline #687: [depth_image_proc] Remapping CameraInfo topics Correctly

swiz23, 2021-07-26:
In the 'Execution' section of the [depth_image_proc](https://wiki.ros.org/depth_image_proc) ROS Wiki, it shows that the camera_info topic can be remapped to a different topic. However, if you take a look at the [nodelet code](https://github.com/ros-perception/image_pipeline/blob/71d537a50bf4e7769af513f4a3d3df0d188cfa05/depth_image_proc/src/nodelets/point_cloud_xyz.cpp#L97), the 'camera_info' topic name is automatically determined by the 'image_rect' topic name. The code there calls the 'subscribeCamera' function that's part of the ImageTransport class. If you look [there](http://docs.ros.org/en/diamondback/api/image_transport/html/classimage__transport_1_1ImageTransport.html#a6754562b0ffe99b0cf716e621d2cfa6b), you'll find text that says

> Subscribe to a synchronized image & camera info topic pair, version for arbitrary boost::function object.

> This version assumes the standard topic naming scheme, where the info topic is named "camera_info" in the same namespace as the base image topic.

>Definition at line 70 of file image_transport.cpp.

So this means that once 'image_rect' is remapped (lets say to '/camera/depth/image_rect_raw' as specified in the Wiki page), then the 'camera_info topic name is *automatically* remapped to /camera/depth/camera_info without having to do it manually. So the camera_info remapping line should be deleted.

However, let's say that for whatever reason, your camera_info topic was actually called /camera/camera_info instead (which goes against the standard - but let's just say that), then...

```
<launch>
  <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager" />

  <node pkg="nodelet" type="nodelet" name="nodelet1"
        args="load depth_image_proc/point_cloud_xyz nodelet_manager">
    <remap from="camera_info" to="/camera/camera_info"/>
    <remap from="image_rect" to="/camera/depth/image_rect_raw"/>
    <remap from="points" to="/camera/depth/points"/>
  </node>
</launch>
```
would not work at all. Instead, you would have to do...

```
<launch>
  <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager" />

  <node pkg="nodelet" type="nodelet" name="nodelet1"
        args="load depth_image_proc/point_cloud_xyz nodelet_manager">
    <remap from="/camera/depth/camera_info" to="/camera/camera_info"/>
    <remap from="image_rect" to="/camera/depth/image_rect_raw"/>
    <remap from="points" to="/camera/depth/points"/>
  </node>
</launch>
```

The reason I brought this up is because I've actually worked with a camera that does not follow this ROS standard and when I tried using the point_cloud_xyz nodelet, it did not work. Only after doing the remapping as specified in the example above did things work properly.

Marwan-Refaat, 2023-03-02:
I was trying to run this nodelet manager for hours using a rosbag where the camera topics weren't conventionally named and I was so confused as to why it wasn't working. Did the remapping and the issue is fixed! Thank you!

mikeferguson, 2024-02-18:
I don't think there is actually a way to improve this - just because of how remapping and image_pipeline work - but it is documented in ROS2: http://docs.ros.org/en/rolling/p/image_pipeline/tutorials.html#remapping-camera-info-topics