ros-perception/image_pipeline #690: Availability on ROS2 Foxy

Enes1097, 2021-08-11:
Hello,
I was just about to build this package in ROS2 Foxy, but realized that there is no branch referencing the Foxy distribution yet.
Is this package also going to be released in this distribution?

Thank you for your answer!

JWhitleyWork, 2021-08-11:
Version 2.2.1 was released for Foxy and is available as binaries on the ROS repositories. This release was created from the `ros2` branch. However, now that Galactic is out, I will create a `foxy` branch and release for Galactic from the `ros2` branch.

closed 2021-08-11 by JWhitleyWork

Enes1097, 2021-08-12:
Thank you for your quick answer!
Before I asked this question I already tried building the package thinking that the "ROS2" branch is refering to the foxy distribution, but I did not work. This is what my shell gave as an output.

[image]
[image]

How is the package refering to a _catkin_ file, if the package is in ROS2 and not ROS1?

(I'm sorry, I just started working with ROS, so I'm a beginner)

Enes1097, 2021-08-12:
Thank you for your quick answer!
Before I asked this question I already tried building the package thinking that the "ROS2" branch is refering to the foxy distribution, but I did not work. This is what my shell gave as an output.

[image]
[image]

How is the package refering to a catkin file, if the package is in ROS2 and not ROS1?

(I'm sorry, I just started working with ROS, so I'm a beginner)