ros-perception/image_pipeline #695: ROS Camera Calibration for Webcam - Not working

diwamanic, 2021-08-23:
Hi peeps! As a part of learning, I am trying to perform Camera calibration with my laptop's Webcam. I am really new to this process. I will explain everything about how far I have come so far, and where I have issues in coming to an successful calibration.

My system specifications: Ubuntu 18.04, ROS Melodic.

For Camera streaming, I am using usb_cam ros package http://wiki.ros.org/usb_cam. And it is working perfectly fine.

After I launch the usb_cam-test.launch, I looked into the rostopic list, where I could find the following two topics: /usb_cam/image_raw /usb_cam/camera_info

So, I progressed forward to perform the camera calibration with the ros package "camera_calibration", but instead of getting a camera stream with calibration options as given in the tutorials, I get a black screen like below:

https://user-images.githubusercontent.com/22639337/130369027-429f2091-f2b7-466f-a6f5-b0d2725f55c8.png

This calibration task is crucial to perform a task at my work. So, I would really need to get some help from someone who has been through this issue before. Thank you very much for the help.

JWhitleyWork, 2021-08-23:
@diwamanic Did you verify that the topics between `usb_cam` and `camera_calibration` are hooked up to eachother? ROS doesn't do this on it's own - you have to connect the topics using [topic remapping](https://roboticsbackend.com/ros-topic-remap-example/).

diwamanic, 2021-08-24:
Thank you for your answer. I understood my mistake. When I download the image_pipeline, I forgot to clone it with the suitable devel-branch. As I am using melodic version of ROS, I should git clone it with -b indigo-devel as it is mentioned in ROS Wiki.
Thank you for your input though. :)