ros-perception/image_pipeline #700: depth_image_pro publish rate is very low

vishalleo1, 2021-09-08:
we have use the intel camera the inputs rate are the 28 fps, the opute of the depth_registered/camera_info rate is 0.31 fps.
we have use the depth_image_proc ros package for this.
we have use this command 
rosrun nodelet nodelet manager __name:=depth_image_proc

then for register depth  image
rosrun nodelet nodelet load depth_image_proc/register depth_image_proc __name:=nodelet1 rgb/camera_info:=/camera/color/camera_info depth/camera_info:=/camera/depth/camera_info depth/image_rect:=/camera/depth/image_rect_raw

thank for help us
images for the publish rate of image 
[image]
[image]

lucasw, 2021-09-09:
I had low depth registered output rate issues but it was due to low target camera_info input rates combined with an approximate sync policy in the register node- I think the register nodelet should publish at the rate of the input depth image and only require the transform to the target/output frame is valid, let downstream nodes decide if a depth image that isn't synchronized to the target camera info is useful or not.

do335, 2023-04-28:
having the same problem(using ROS2 HUMBLE)

mikeferguson, 2024-01-19:
This appears to be a configuration issue - so I suggest using https://robotics.stackexchange.com to try to debug it

closed 2024-01-19 by mikeferguson