ros-perception/image_pipeline #721: Make depth_image_proc fill_upsampling_holes work with full range of relative angles

lucasw, 2022-01-16:
The current implementation doesn't draw anything from certain relative angles between the source depth image frame and target frame (called rgb in the code and documentation but it is just any valid camera info with a valid frame), draws increasingly thin slivers as it approaches those angles.

I have the beginning of a quick fix here https://github.com/lucasw/image_pipeline/commit/478fd5bf4ac5a0229e33e1ca7ff7c8e226f00124 but it'll have to do 3 or 4x the computation.  It's not the full triangle rasterization mentioned in a `todo` but would get decent results in more cases.  It gets a little ridiculous to embed what looks more and more like a full software 3D renderer here into this node but making it a little better seems reasonable.

This image shows off the filled areas getting reduced to slivers using the min/max of two values approach in the commit above, with it the slivers can get reduced to nothing at all because the range check can fail entirely.
[image]

This is the original code with for loops that sometimes don't execute at all:
https://github.com/ros-perception/image_pipeline/blob/noetic/depth_image_proc/src/nodelets/register.cpp#L291-L293

There look to be some other obvious refactors/cleanups to do in that same code (it's doing the same thing in three nearly identical blocks).  Also I can look at creating a unit test for this (or is there one somewhere and I'm not seeing it?).

Probably few people who are using this have crazy high angles between the depth camera and the other camera frame, I'm only running into it now trying this out with synthetic data (I may get some of the synthetic data generation into the unit test, but it's nice to have manual test support also).

flynneva, 2022-01-24:
@lucasw do you think you could also port this to the `ros2` branch of `depth_image_proc`? I did a pretty major rewrite of that package awhile ago so it might be a bit easier to add it there.

lucasw, 2022-01-24:
@flynneva I'll take a look, but would probably be slow to get that done.

flynneva, 2022-01-24:
@lucasw I could take a stab at it later this week and tag you for a review 👍🏼 after looking through your updates I think it'll fit in nicely with the rewrite i did (lots of consolidation and templating)