ros-perception/image_pipeline #725: Undefines variable implying a crash when calibrating

teyssieuman, 2022-01-30:
'fisheye_flags' is not defined here.

https://github.com/ros-perception/image_pipeline/blob/1b6f868501ee0a753a3138bf31efd57c4054c5dc/camera_calibration/src/camera_calibration/camera_calibrator.py#L108

This make the calibration fails :

```
 ros2 run camera_calibration cameracalibrator -p chessboard --size 7x10 -q 0.016 image:=/image_raw camera:=/camera_info
[WARN] [1643563428.060152243] [rcl]: Found remap rule 'image:=/image_raw'. This syntax is deprecated. Use '--ros-args --remap image:=/image_raw' instead.
[WARN] [1643563428.060167887] [rcl]: Found remap rule 'camera:=/camera_info'. This syntax is deprecated. Use '--ros-args --remap camera:=/camera_info' instead.
[WARN] [1643563428.063036858] [rcl]: Found remap rule 'image:=/image_raw'. This syntax is deprecated. Use '--ros-args --remap image:=/image_raw' instead.
[WARN] [1643563428.063044828] [rcl]: Found remap rule 'camera:=/camera_info'. This syntax is deprecated. Use '--ros-args --remap camera:=/camera_info' instead.
Waiting for service camera/set_camera_info ...
OK
Waiting for service left_camera/set_camera_info ...
OK
Waiting for service right_camera/set_camera_info ...
OK
Traceback (most recent call last):
  File "/home/sdri1/Bureau/sdria_loop_ws/install/lib/camera_calibration/cameracalibrator", line 33, in <module>
    sys.exit(load_entry_point('camera-calibration', 'console_scripts', 'cameracalibrator')())
  File "/home/sdri1/Bureau/sdria_loop_ws/build/camera_calibration/src/camera_calibration/nodes/cameracalibrator.py", line 144, in main
    node = OpenCVCalibrationNode("cameracalibrator", boards, options.service_check, sync, calib_flags, pattern, options.camera_name,
  File "/home/sdri1/Bureau/sdria_loop_ws/build/camera_calibration/src/camera_calibration/camera_calibrator.py", line 223, in __init__
    CalibrationNode.__init__(self, *args, **kwargs)
  File "/home/sdri1/Bureau/sdria_loop_ws/build/camera_calibration/src/camera_calibration/camera_calibrator.py", line 108, in __init__
    self._fisheye_calib_flags = fisheye_flags
NameError: name 'fisheye_flags' is not defined
```

JWhitleyWork, 2022-01-31:
Which branch/version is this on? You linked to a specific commit so it's hard for me to tell.

KiloNovemberDelta, 2022-02-08:
At this instruction : [        self._fisheye_calib_flags = fisheye_flags](https://github.com/ros-perception/image_pipeline/blob/1b6f868501ee0a753a3138bf31efd57c4054c5dc/camera_calibration/src/camera_calibration/camera_calibrator.py#L108)

I have just set manually  : self._fisheye_calib_flags = False 
And then compile again. Indeed, it is a bug to fix.

mikeferguson, 2024-01-19:
Assigning myself - going to be looking over calibration stuff in the next two weeks

mikeferguson, 2024-02-01:
https://github.com/ros-perception/image_pipeline/commit/2640ed1e0a9ad513b88ec4010b0b4fd857fb3ff0 has added fisheye_flags to the constructor parameters