ros-perception/image_pipeline #727: Difference between image_raw and image_color outputs.

gustavojoseleite, 2022-02-04:
The documentation say:
image_raw - Raw image stream from the camera driver. 
image_color - Color unrectified image. 
So if image_color is unrectified, what is the exactly difference between image_raw and image_color? When i should be use image_color rather than image_raw?

Thanks

lucasw, 2022-02-05:
The image_raw could have color information in it but in a bayer pattern, then image_color is processed (by image_proc) to interpolate pixel values into rgb/bgr format.  That converted color image takes up 3x the bytes as the raw image if the input was 8bpp.

gustavojoseleite, 2022-02-06:
Right, so the image_color has some processing for the pattern of color, for example, but doesn't have information about camera parameters just the image_color_rect have this correction, right?

Thanks for your reply @lucasw .

closed 2024-01-19 by mikeferguson