ros-perception/image_pipeline #737: fisheye calibration fails

arjunskumar, 2022-03-11:
Hi, I'm not getting proper results with fisheye mode calibration

TobiMiller, 2022-03-28:
```Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/camera_calibrator.py", line 269, in on_mouse
    self.c.do_calibration()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/calibrator.py", line 872, in do_calibration
    print((self.ost()))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/calibrator.py", line 758, in ost
    return self.lrost(self.name, self.distortion, self.intrinsics, self.R, self.P, self.size)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/camera_calibration/calibrator.py", line 534, in lrost
    assert len(calmessage) < 525, "Calibration info must be less than 525 bytes"
AssertionError: Calibration info must be less than 525 bytes
```

Thats the problem i am facing with melodic.

arjunskumar, 2022-03-29:
@TobiMiller with ros noetic it is working

mikeferguson, 2024-01-17:
Appears fixed in noetic - and melodic is EOL

closed 2024-01-17 by mikeferguson