ros-perception/image_pipeline #739: CropDecimateNode Number Subscribers / image_view

mhubii, 2022-03-27:
Hi there, the `CropDecimateNode` won't process subscribed images unless the number of subscriber is > 0, see

https://github.com/ros-perception/image_pipeline/blob/3cc137863ab039aa1a1a2271e6df2d9622479089/image_proc/src/crop_decimate.cpp#L134

intuitively, this makes sense. However, if one subscribes via `ros2 run image_view image_view --ros-args --remap image:=/out/image_raw`, the number of subscribers remains 0. 

I don't know if this is a bug of the `image_view` node, or desired behavior

JWhitleyWork, 2022-03-27:
This is desired behavior to save the CPU cycles if there are no subscribers. With regard to `image_view`, I have seen this behavior before and I'm not exactly sure what causes it. If you find any sort of cause/fix, feel free to open a PR.

closed 2022-03-27 by JWhitleyWork

mhubii, 2022-03-27:
thanks for the prompt response. This is no particular issue of `image_view`, but rather of `image_transport`. I created a minimal example to reproduce, see ros-perception/image_common#238