ros-perception/image_pipeline #740: foxy version: image_rotate does not initialize the target and source _vectors in constructor

dcconner, 2022-04-04:
Need to add the following after onInit() call to update data; target_vector and source_vector default to zero

` 
  target_vector_.vector.x = config_.target_x;
  target_vector_.vector.y = config_.target_y;
  target_vector_.vector.z = config_.target_z;

  source_vector_.vector.x = config_.source_x;
  source_vector_.vector.y = config_.source_y;
  source_vector_.vector.z = config_.source_z;

`

clalancette, 2022-04-26:
Can you please open a pull request with that change so we can review it?  Thanks.

dcconner, 2022-05-01:
I verified compilation for image_rotate, but I went in another direction and developed a new node for ROS 2 I called image_flip that I present in PR #743 for review .  I'm not set up to fully test this PR

PR #862 (2024-01-19): image_rotate: clean up
This is the first component/node with a cleanup pass to be fully implemented:

 * Fix #740 by initializing vectors. Do this by declaring parameters AFTER we define the callback
 * Implement lazy subscribers (I missed this in the earlier PRs)
 * Add image_transport parameter so we can specify that desired transport of our subscriptions
 * Update how we test for connectivity (and update the debug message - has nothing to do with whether we are remapped, it's really about whether we are connected)
commit 21e5f25a (2024-01-19): image_rotate: clean up (#862)

This is the first component/node with a cleanup pass to be fully
implemented:

* Fix #740 by initializing vectors. Do this by declaring parameters
AFTER we define the callback
 * Implement lazy subscribers (I missed this in the earlier PRs)
* Add image_transport parameter so we can specify that desired transport
of our subscriptions
* Update how we test for connectivity (and update the debug message -
has nothing to do with whether we are remapped, it's really about
whether we are connected)
closed 2024-01-19 by mikeferguson