ros-perception/image_pipeline #748: ROS2 : stereo_image_proc : rectification/undistortion

felixnil, 2022-05-13:
Hi,
As far as I undestand, in the ROS1 version of stereo_image_proc package, the stereo_image_proc node allows to transform the raw images of a stereo pair (+ cameraInfo) into the undistorted and rectified ones.

In the ROS2 version of the package however, the stereo_image_proc node has disappeared from the stereo_image_proc package (the nodes for disparity and for point clouds still exist).

So is there now a new way in ROS2 to convert raw_images to undistorted images with stereo rectification?
Or is it just that the stereo_image_proc is not ported yet? In this case, is there already someone working on it?

Thanks a lot in advance
Best Regards
Felix

JWhitleyWork, 2022-07-13:
It looks like this is supposed to be solved by adding the `launch_image_proc` argument [here](https://github.com/ros-perception/image_pipeline/blob/ea216eba6d2c611adbd78bb9ef9db25816d1405c/stereo_image_proc/launch/stereo_image_proc.launch.py#L122) which will launch individual debayer and rectify nodes for each camera. However, based on [this PR](https://github.com/ros-perception/image_pipeline/pull/752), that doesn't appear to work correctly in this launch file. That being said, until this is fixed on the `ros2` branch, you should be able to launch these nodes yourself and do the rectification on each individual camera before passing the images to either the Point Cloud or Disparity nodes.

mikeferguson, 2024-02-11:
There have been numerous fixes to this launch file since 2022 - and based on my review, it appears that everything is working except some image_transport (which I ticketed here: https://github.com/ros-perception/image_pipeline/issues/940)

closed 2024-02-11 by mikeferguson