ros-perception/image_pipeline #753: ROS2 Galactic: Model change trackbar missing on GUI

rafaelRinconFo, 2022-05-31:
Currently, I have the Ros2 galactic installation and I'm trying to calibrate a Fisheye camera. After checking for a while I realized that the Trackbar for switching between pinhole and fisheye cameras should be displayed on the GUI as stated in [here](https://github.com/ros-perception/image_pipeline/blob/3cc137863ab039aa1a1a2271e6df2d9622479089/camera_calibration/src/camera_calibration/camera_calibrator.py#L245) but that's not the case as you can see below.

[image]

Is there any other way to calibrate a fisheye camera like an argument or parameter in the `ros2 run camera_calibration cameracalibrator` command while it gets fixed?

JWhitleyWork, 2022-07-13:
@rafaelRinconFo From the size of this window and the objects in it, it appears that the resolution of the screen that you are using is very small. This would cause some elements in the Python window to not be visible. Can you please confirm your screen resolution? Thanks.

rafaelRinconFo, 2022-07-18:
@JWhitleyWork I'm afraid that's not the case. The screen I'm using has a 1920x1080 resolution. In addition, I had to modify a little bit of the code for making the trackbar appear, so, I think the problem is not related to the resolution.

mikeferguson, 2024-01-19:
Assigning myself to look at this one - will be reviewing calibration stuff at some point in the next two weeks

mikeferguson, 2024-01-31:
Galactic is no longer supported - but I have confirmed that the model selection shows up on rolling (even if the window is very small - it's the other buttons that get cut off). I'm thinking maybe that feature never got ported to Galactic?

closed 2024-01-31 by mikeferguson