ros-perception/image_pipeline #765: Disparity maps are so bad for stereo cameras

farhad-dalirani, 2022-07-24:
Hello, 

These images are from my left and right cameras. The third image is a disparity map.

I generated it with these commands:

```
ROS_NAMESPACE=ros_basler_stereo rosrun stereo_image_proc stereo_image_proc _approximate_sync:=True
```
and
```
rosrun image_view stereo_view stereo:=/ros_basler_stereo image:=image_raw _approximate_sync:=True _queue_size:=800
```
[image]

lidarmansiwon, 2023-02-01:
I have same lssue. did you solve that issue? 
[image]

farhad-dalirani, 2023-02-01:
There can be several methods, three of them:

1- change the parameter for creating a depth map
2- change the distance between the two cameras. For different depths, different baselines obtain different results. For example, inside a small room, a baseline of 10 cm can be enough. However, for cars on the streets, it can be 20 cm.
3- Maybe your environment does not have enough features. Therefore, the matching fails.

lidarmansiwon, 2023-02-02:
> There can be several methods, three of them:
>
> 1- change the parameter for creating a depth map 2- change the distance between the two cameras. For different depths, different baselines obtain [...]

Thank you for your quick reply!
Can i use [rosrun rqt_reconfigure rqt_reconfigure] to change the parameter?? how can i change parameter for a depth map..?

mikeferguson, 2024-01-19:
This doesn't appear to really be an actionable ticket - all of the suggestions are about the configuration and hardware setup rather than some intrinsic problem in the code.

closed 2024-01-19 by mikeferguson