ros-perception/image_pipeline #775: Can't compile image_rotate for ros_rolling

Kaju-Bubanja, 2022-09-20:
I am compiling everything for ros_rolling currently and when trying to compile this package on the rolling branch I get following error trying to compile the image_transport package:
```

FAILED: CMakeFiles/image_rotate.dir/src/image_rotate_node.cpp.o 
/usr/bin/c++ -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DRCUTILS_ENABLE_FAULT_INJECTION -Dimage_rotate_EXPORTS -I/home/username/ros2_rolling/src/ros-perception/image_pipeline/image_rotate/include -isystem /home/username/ros2_rolling/install/tf2_geometry_msgs/include/tf2_geometry_msgs -isystem /home/username/ros2_rolling/install/tf2_ros/include/tf2_ros -isystem /home/username/ros2_rolling/install/tf2/include/tf2 -isystem /home/username/ros2_rolling/install/image_transport/include/image_transport -isystem /home/username/ros2_rolling/install/cv_bridge/include/cv_bridge -isystem /home/username/ros2_rolling/install/sensor_msgs/include/sensor_msgs -isystem /home/username/ros2_rolling/install/rclcpp_components/include/rclcpp_components -isystem /home/username/ros2_rolling/install/rclcpp/include/rclcpp -isystem /home/username/ros2_rolling/install/geometry_msgs/include/geometry_msgs -isystem /home/username/ros2_rolling/install/rcl_interfausername/include/rcl_interfaces -isystem /home/username/ros2_rolling/install/class_loader/include/class_loader -isystem /home/username/ros2_rolling/install/rcpputils/include/rcpputils -isystem /home/username/ros2_rolling/install/rcutils/include/rcutils -isystem /home/username/ros2_rolling/install/builtin_interfausername/include/builtin_interfaces -isystem /home/username/ros2_rolling/install/rosidl_runtime_c/include/rosidl_runtime_c -isystem /home/username/ros2_rolling/install/rosidl_typesupport_interface/include/rosidl_typesupport_interface -isystem /home/username/ros2_rolling/install/std_msgs/include/std_msgs -isystem /home/username/ros2_rolling/install/fastcdr/include -isystem /home/username/ros2_rolling/install/rosidl_runtime_cpp/include/rosidl_runtime_cpp -isystem /home/username/ros2_rolling/install/rosidl_typesupport_fastrtps_cpp/include/rosidl_typesupport_fastrtps_cpp -isystem /home/username/ros2_rolling/install/rmw/include/rmw -isystem /home/username/ros2_rolling/install/rosidl_typesupport_fastrtps_c/include/rosidl_typesupport_fastrtps_c -isystem /home/username/ros2_rolling/install/rosidl_typesupport_introspection_c/include/rosidl_typesupport_introspection_c -isystem /home/username/ros2_rolling/install/rosidl_typesupport_introspection_cpp/include/rosidl_typesupport_introspection_cpp -isystem /usr/include/opencv4 -isystem /home/username/ros2_rolling/install/ament_index_cpp/include/ament_index_cpp -isystem /home/username/ros2_rolling/install/libstatistics_collector/include/libstatistics_collector -isystem /home/username/ros2_rolling/install/rcl/include/rcl -isystem /home/username/ros2_rolling/install/rcl_logging_interface/include/rcl_logging_interface -isystem /home/username/ros2_rolling/install/rcl_yaml_param_parser/include/rcl_yaml_param_parser -isystem /home/username/ros2_rolling/install/libyaml_vendor/include/libyaml_vendor -isystem /home/username/ros2_rolling/install/tracetools/include/tracetools -isystem /home/username/ros2_rolling/install/statistics_msgs/include/statistics_msgs -isystem /home/username/ros2_rolling/install/rosgraph_msgs/include/rosgraph_msgs -isystem /home/username/ros2_rolling/install/rosidl_typesupport_cpp/include/rosidl_typesupport_cpp -isystem /home/username/ros2_rolling/install/rosidl_typesupport_c/include/rosidl_typesupport_c -isystem /home/username/ros2_rolling/install/message_filters/include/message_filters -isystem /home/username/ros2_rolling/install/composition_interfausername/include/composition_interfaces -isystem /home/username/ros2_rolling/install/orocos_kdl_vendor/include/orocos_kdl -isystem /usr/include/eigen3 -isystem /home/username/ros2_rolling/install/rclcpp_action/include/rclcpp_action -isystem /home/username/ros2_rolling/install/action_msgs/include/action_msgs -isystem /home/username/ros2_rolling/install/unique_identifier_msgs/include/unique_identifier_msgs -isystem /home/username/ros2_rolling/install/rcl_action/include/rcl_action -isystem /home/username/ros2_rolling/install/tf2_msgs/include/tf2_msgs -g -fPIC -Wall -Wextra -Wpedantic -std=gnu++17 -MD -MT CMakeFiles/image_rotate.dir/src/image_rotate_node.cpp.o -MF CMakeFiles/image_rotate.dir/src/image_rotate_node.cpp.o.d -o CMakeFiles/image_rotate.dir/src/image_rotate_node.cpp.o -c /home/username/ros2_rolling/src/ros-perception/image_pipeline/image_rotate/src/image_rotate_node.cpp
In file included from /home/username/ros2_rolling/src/ros-perception/image_pipeline/image_rotate/src/image_rotate_node.cpp:50:
/home/username/ros2_rolling/install/cv_bridge/include/cv_bridge/cv_bridge/cv_bridge.h:42:2: warning: #warning This header is obsolete, please include cv_bridge/cv_bridge.hpp instead [-Wcpp]
   42 | #warning This header is obsolete, please include cv_bridge/cv_bridge.hpp instead
      |  ^~~~~~~
In file included from /home/username/ros2_rolling/src/ros-perception/image_pipeline/image_rotate/include/image_rotate/image_rotate_node.hpp:41,
                 from /home/username/ros2_rolling/src/ros-perception/image_pipeline/image_rotate/src/image_rotate_node.cpp:42:
/home/username/ros2_rolling/install/tf2_ros/include/tf2_ros/tf2_ros/transform_broadcaster.h: In instantiation of ‘tf2_ros::TransformBroadcaster::TransformBroadcaster(NodeT&&, const rclcpp::QoS&, const rclcpp::PublisherOptionsWithAllocator<AllocatorT>&) [with NodeT = image_rotate::ImageRotateNode&; AllocatorT = std::allocator<void>]’:
/usr/include/c++/11/ext/new_allocator.h:162:4:   required from ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = tf2_ros::TransformBroadcaster; _Args = {image_rotate::ImageRotateNode&}; _Tp = tf2_ros::TransformBroadcaster]’
/usr/include/c++/11/bits/alloc_traits.h:516:17:   required from ‘static void std::allocator_traits<std::allocator<_Tp1> >::construct(std::allocator_traits<std::allocator<_Tp1> >::allocator_type&, _Up*, _Args&& ...) [with _Up = tf2_ros::TransformBroadcaster; _Args = {image_rotate::ImageRotateNode&}; _Tp = tf2_ros::TransformBroadcaster; std::allocator_traits<std::allocator<_Tp1> >::allocator_type = std::allocator<tf2_ros::TransformBroadcaster>]’
/usr/include/c++/11/bits/shared_ptr_base.h:519:39:   required from ‘std::_Sp_counted_ptr_inplace<_Tp, _Alloc, _Lp>::_Sp_counted_ptr_inplace(_Alloc, _Args&& ...) [with _Args = {image_rotate::ImageRotateNode&}; _Tp = tf2_ros::TransformBroadcaster; _Alloc = std::allocator<tf2_ros::TransformBroadcaster>; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/11/bits/shared_ptr_base.h:650:16:   required from ‘std::__shared_count<_Lp>::__shared_count(_Tp*&, std::_Sp_alloc_shared_tag<_Alloc>, _Args&& ...) [with _Tp = tf2_ros::TransformBroadcaster; _Alloc = std::allocator<tf2_ros::TransformBroadcaster>; _Args = {image_rotate::ImageRotateNode&}; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/11/bits/shared_ptr_base.h:1342:14:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<tf2_ros::TransformBroadcaster>; _Args = {image_rotate::ImageRotateNode&}; _Tp = tf2_ros::TransformBroadcaster; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
/usr/include/c++/11/bits/shared_ptr.h:409:59:   required from ‘std::shared_ptr<_Tp>::shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<tf2_ros::TransformBroadcaster>; _Args = {image_rotate::ImageRotateNode&}; _Tp = tf2_ros::TransformBroadcaster]’
/usr/include/c++/11/bits/shared_ptr.h:862:14:   required from ‘std::shared_ptr<_Tp> std::allocate_shared(const _Alloc&, _Args&& ...) [with _Tp = tf2_ros::TransformBroadcaster; _Alloc = std::allocator<tf2_ros::TransformBroadcaster>; _Args = {image_rotate::ImageRotateNode&}]’
/usr/include/c++/11/bits/shared_ptr.h:878:39:   required from ‘std::shared_ptr<_Tp> std::make_shared(_Args&& ...) [with _Tp = tf2_ros::TransformBroadcaster; _Args = {image_rotate::ImageRotateNode&}]’
/home/username/ros2_rolling/src/ros-perception/image_pipeline/image_rotate/src/image_rotate_node.cpp:360:60:   required from here
/home/username/ros2_rolling/install/tf2_ros/include/tf2_ros/tf2_ros/transform_broadcaster.h:71:7: error: base operand of ‘->’ has non-pointer type ‘image_rotate::ImageRotateNode’
   71 |       node->get_node_parameters_interface(),
      |       ^~~~
ninja: build stopped: subcommand failed.

```

Is this an issues with my system and you can compile this package for rolling or is this a bug in the package?

ijnek, 2022-09-20:
Wih the warning ``#warning This header is obsolete, please include cv_bridge/cv_bridge.hpp instead``, that is due to #include a header file that has been deprecated, and so it is unrelated the error below that.

ijnek, 2022-09-20:
In regards to the error, I believe this is a bug in tf2_ros, and so I've opened a [ticket on ros2/geometry2](https://github.com/ros2/geometry2/issues/554)

mikeferguson, 2024-01-17:
CI is passing at this point