ros-perception/image_pipeline #779: ros2 foxy image_view extract_images transport param always is --ros-args_sub

lucasrj, 2022-10-06:
then i tray to use the extractor in my luanchfile i allwais get an error 

'what():  Unable to load plugin for transport 'image_transport/--ros-args_sub', error string:'

but if change name of transport to test in the code works

[launch.log](https://github.com/ros-perception/image_pipeline/files/9724773/launch.log)
[launch.log](https://github.com/ros-perception/image_pipeline/files/9724779/launch.log)
[image_extractor.launch](https://github.com/ros-perception/image_pipeline/files/9724786/image_extractor.launch.txt)

mikeferguson, 2024-01-17:
Possibly same cause as https://github.com/ros-perception/image_pipeline/issues/824?

mikeferguson, 2024-01-19:
Foxy is EOL - and image_transport has had inconsistent support - support is now consistently addressed in rolling and will be part of Jazzy

closed 2024-01-19 by mikeferguson