ros-perception/image_pipeline #780: [ROS2] image_proc: replace SubscriberStatusCallback (from ROS1) with a Timer

AndreasR30, 2022-10-06:
In ROS1 the `image_proc` nodelets only subscribed to the input topic when there was an [subscription on their output topic](https://github.com/ros-perception/image_pipeline/blob/noetic/image_proc/src/nodelets/debayer.cpp#L104).

In ROS2 the components always subscribe to their input topic. For one component this is not a big problem because in the message callback it is [checked](https://github.com/ros-perception/image_pipeline/blob/rolling/image_proc/src/debayer.cpp#L106) with `getNumSubscribers()` whether there are consuming nodes/components. If there is no subscriber then no work is done.
However, when there are multiple components in the pipeline (e.g. debayer -> rectify), then there will be always a consuming node from the perspective of the debayer component and it will do expensive image processing even though nobody is interested in the output of the image pipeline.

Since there are no such callbacks for `image_transport::Publisher` or for` rclcpp::Publisher`, which is internally used, and it does not seem to get such a callback in the near future, I would suggest that the only solution is to check with a timer whether `getNumSubscribers() > 0`. If so, the node subscribes to its input topic. Otherwise it will unsubscribe from the input topic.

Maybe part of #467

AndreasR30, 2022-10-06:
Related:
- Feature request for lazy `image_transport::subscriber/publisher`: ros-perception/image_common#111
- This would require low level support in RMW: ros2/rmw#155

mikeferguson, 2023-05-17:
I'm just starting to work on this for Iron (but not with a timer - using the newly exposed matched event). I've got CropDecimate working as an example so far, should have a PR in a few days for all the components. Final goal is to be able to do https://github.com/ros-drivers/openni2_camera/issues/119 (and then make rgbd_launch work)

mikeferguson, 2024-01-18:
Lazy subscribers are now in rolling!