ros-perception/image_pipeline #781: camera calibration module conflicts with python version "ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)"

nagarjunvinukonda, 2022-10-20:
Hi,

I am having trouble when I run the camera calibration module with ROS melodic. Ubuntu 18.0.4. as follows:

```
rrcam@rrcam20220901-001:~/catkin_ws$ rosrun camera_calibration cameracalibrator.py --size 12x8 --square 0.05 image:=/cam1
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/rrcam/catkin_ws/src/camera_calibration/src/camera_calibration/camera_calibrator.py", line 118, in run
    self.function(m)
  File "/home/rrcam/catkin_ws/src/camera_calibration/src/camera_calibration/camera_calibrator.py", line 242, in handle_monocular
    max_chessboard_speed = self._max_chessboard_speed)
  File "/home/rrcam/catkin_ws/src/camera_calibration/src/camera_calibration/calibrator.py", line 954, in __init__
    super(MonoCalibrator, self).__init__(*args, **kwargs)
TypeError: super() argument 1 must be type, not classobj
```

Hence I installed python3 version of ros dependencies:
```
 apt install python3-pip python3-all-dev python3-rospkg
 apt install ros-melodic-desktop-full --fix-missing
 sudo apt install python-rosdep
```
 
But still not change I am having import error:
```
rrcam@rrcam20220901-001:~/catkin_ws$ rosrun camera_calibration cameracalibrator.py --size 12x8 --square 0.05 image:=/cam1
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/rrcam/catkin_ws/src/camera_calibration/src/camera_calibration/camera_calibrator.py", line 118, in run
    self.function(m)
  File "/home/rrcam/catkin_ws/src/camera_calibration/src/camera_calibration/camera_calibrator.py", line 255, in handle_monocular
    drawable = self.c.handle_msg(msg,self.pub)
  File "/home/rrcam/catkin_ws/src/camera_calibration/src/camera_calibration/calibrator.py", line 1184, in handle_msg
    gray = self.mkgray(msg)
  File "/home/rrcam/catkin_ws/src/camera_calibration/src/camera_calibration/calibrator.py", line 386, in mkgray
    if self.br.encoding_to_dtype_with_channels(msg.encoding)[0] in ['uint16', 'int16']:
  File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)
```

What is the solution for this? why I am getting this error?

aristow1, 2022-11-16:
try builduing your package using the default python 2.7 in Ubuntu 18.

MARHEM602, 2023-01-11:
J'ai le même problème ...

mikeferguson, 2024-01-17:
Melodic/18.04 are now EOL

closed 2024-01-17 by mikeferguson