ros-perception/image_pipeline #796: image_view/camara_calibration cannot display momo16

han88, 2023-01-01:
Hi Im trying to calibrate my orbbec astra mini depth camera running with this  driver (https://github.com/orbbec/ros_astra_camera). The rgb calibration worked without problems, but for the ir stream the image_view/calibration window just show a black display without any error messages. camera/ir/image_raw  topic encoding is mono16! In rviz i can see the ir stream...
Thanks

han88, 2023-01-03:
managed the depth calibration, wasn't the issue of display error in image_view...

closed 2023-01-03 by han88

foxydevelby, 2023-04-26:
> managed the depth calibration, wasn't the issue of display error in image_view...

Hi @han88, how did you resolved your issue, I have the same. no errors but the qt window cannot display the IR stream (mono16)

han88, 2023-05-03:
Hi I had that issue while calibration of the astra mini camera...there is
now an tutorial in the readme file of the ros_asta_carmera package..

Seifert