ros-perception/image_pipeline #802: image_publisher ignores filename and camera_info_url parameters

hpoirier, 2023-02-09:
When starting the image_publisher node, the filename parameter is ignored and the camera_info_url can only be set after the node is up using the ros2 set param command. Moreover filename is not declared in the image_publisher component but only in the image_publisher_node application which makes it impossible to use as a ComposableNode.
This requires some changes in the way parameters are handle when the node is created and in the on_set_parameters_callback.

mikeferguson, 2024-01-19:
The node part works with both parameter or command line arg:
```
ros2 run image_publisher image_publisher_node --ros-args -p filename:=test.png
ros2 run image_publisher image_publisher_node test.png
```
It does appear that filename is not properly defined when using the component alone, and also that the callback is not called on startup when using the component - should be fixed by https://github.com/ros-perception/image_pipeline/pull/861

mikeferguson, 2024-01-19:
PR merged

closed 2024-01-19 by mikeferguson