ros-perception/image_pipeline #823: No executable for depth_image_proc

sagardhatrak, 2023-06-14:
Not able to find the executables for depth_image_proc. 

Following error is printed on command line:
ros2 launch depth_image_proc convert_metric.launch.py 
[INFO] [launch]: All log files can be found below /home/sd/.ros/log/2023-06-14-12-24-37-298393-sd-virtual-machine-19878
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: Node.__init__() missing 1 required keyword-only argument: 'executable'

mikeferguson, 2023-06-14:
I think the issue here is actually just that the launch file has not been updated and you are on a newer version of ros2 than it was last tested on - try changing "node_executable" on line 73 to "executable"

808brick, 2023-06-18:
I think there is some other underlying problem with the package config, because  `ros2 pkg executables depth_image_proc` yields no results. This is with a fresh install using `apt install ros-humble-image-pipeline`

If I had to take a guess, something is wrong with the way the executables are being created in the CMakeLists. Currently it seems like the `rclcpp_components_register_nodes` call is not working properly:
https://github.com/ros-perception/image_pipeline/blob/humble/depth_image_proc/CMakeLists.txt#L39

versus something like `rclcpp_components_register_node` which seems to be working just fine in the other image_pipeline packages, such as stereo_image_proc:
https://github.com/ros-perception/image_pipeline/blob/humble/stereo_image_proc/CMakeLists.txt#L29

UPDATE: After close inspection, it seems you guys want these image_proc nodes used as containerized nodelets instead of regular node. I'm not sure why this is, as it seems more straight forward to simply have a node to do conversions for you which is easily callable with `ros2 run`. I was able to get a crude example "working" in humble with the following, but this really does seem way more complex than it should be:

```python
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    container = ComposableNodeContainer(
            name='my_container',
            namespace='',
            package='rclcpp_components',
            executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='depth_image_proc',
                    plugin='depth_image_proc::PointCloudXyzNode',
                    name='point_cloud_xyz_node',
                    remappings=[
                        ('/image_rect', '/stereo/depth')
                        ('/camera_info', '/stereo/camera_info')
                    ]
                )
            ],
            output='screen',
    )

    return launch.LaunchDescription([container])
```

Hopefully this is helpful to someone.

mikeferguson, 2024-01-17:
Assigning myself to take a closer look - we should have both components AND executables available

@808brick the reason we want components available (ROS2 version of nodelets) is so that if you compose multiple components in an image pipeline, you don't have the overhead of serialize/deserialize and network transport to pass the messages between the nodes. There is a deep dive on the performance impacts of components in this paper: https://arxiv.org/pdf/2305.09933.pdf

mikeferguson, 2024-01-18:
Ok - diving in here - and yep - depth_image_proc definitely wasn't implemented correctly to allow this. I think image_proc worked once upon a time, but has since broken. https://github.com/ros-perception/image_pipeline/pull/852 fixes both.

mikeferguson, 2024-01-18:
Merged into rolling, working on iron/humble back ports

closed 2024-01-18 by mikeferguson