ros-perception/image_pipeline #824: [ros2] image_view only subscribe to raw

ahcorde, 2023-06-29:


ahcorde, 2023-06-29:
FYI @JWhitleyWork

ahcorde, 2023-06-29:
Thinking a little bit more about the second issue, using this remapping is fixed, maybe we should document this better:

```bash
ros2 run image_view image_view compressed "/image/compressed:=/camera/image/compressed"
```

ahcorde, 2023-06-30:
I created this PR https://github.com/ros-perception/image_pipeline/pull/825

mikeferguson, 2024-01-17:
So is it possible you're just using the wrong command line? If I do the following:

```
ros2 run image_view image_view --ros-args --remap image:=/camera/image -p image_transport:=compressed
```

Which is what is shown in the image_transport_tutorials README, then things work (even using what is in rolling debs right now):

```
ros2 run image_view image_view --ros-args --remap image:=/camera/image -p image_transport:=compressed
[INFO] [1705521253.837407238] [image_view_node]: Using transport "compressed"
```

I think this is because without the "-p" newer ROS 2 versions are interpreting this as a remap rather than a parameter

mikeferguson, 2024-01-17:
There is definitely something funky going on with the remapping - this fully works against current rolling debs:

```
ros2 run image_view image_view --ros-args --remap image/compressed:=/camera/image/compressed -p image_transport:=compressed
```

mikeferguson, 2024-01-17:
Based on these comments - I opened https://github.com/ros-perception/image_pipeline/pull/850 to at least correct the help string for image_view to the proper format

mikeferguson, 2024-01-18:
I believe that https://github.com/ros-perception/image_pipeline/pull/851 is the correct approach to fix the remap issues - still need to apply that to this particular node

closed 2024-01-18 by mikeferguson