ros-perception/image_pipeline #828: Problem with QoS

rafal-gorecki, 2023-07-06:
Hi,
I have run Astra camera on ROS2 Humble and I'd like to save some image with `image_save` node, but when I use service `save` i receive this error:

```
New publisher discovered on topic '/camera/color/image_raw', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```

Camera send image with Best Effort policy, which is recommend for sensor, so I'd rather try to change `image_save` subscrion part.
Could you help me fix this Issue?

mikeferguson, 2024-01-17:
I think the best practice has changed with https://ros.org/reps/rep-2003.html - the camera should actually be publishing Reliable and subscribing BestEffort (which maximizes compatibility).

Regardless, it looks like image_save should switch to best effort

mikeferguson, 2024-01-17:
Now tracking under: https://github.com/ros-perception/image_pipeline/issues/847

closed 2024-01-17 by mikeferguson